MATPOWER/matpower #132: Finding Shift Factor after applying contingencies

thomsonian2023, 2021-11-22:
Is it possible to calculate shift factor from PSSE RAW file using Matpower after applying a list of contingencies i.e. monitored lines which can potentially go on outage?

rdzman, 2021-11-22:
Once you convert the RAW file to MATPOWER format using `psse2mpc()` you can compute line outage distribution factors using [`makeLODF()`](https://matpower.org/docs/ref/matpower7.1/lib/makeLODF.html) (also described in Section 4.5 in the [User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf)). Is that what you want?

thomsonian2023, 2021-11-22:
It's not LODF calculation.

There is base RAW file and it comes with a list of contingencies. i.e. list of lines which could potentially go out but are still in service in the RAW file.

Since, this is a list of hypothetical outages, so they are not represented in RAW file. Hence, RAW file needs this additional outage information before calculating new shift factor accounting for list of contingencies.

thomsonian2023, 2021-11-22:
That will be an additional file probably in a csv format.

closed 2021-11-22 by thomsonian2023