MATPOWER/matpower #136: Power flow calculation with distributed slack bus

richardkue, 2022-01-07:
To the best of my knowledge there is no power flow calculation with a distributed slack bus in matpower. I am thinking about making a contribution in this regard. Is it enough to implement a power flow with distributed slack bus in only one algorithm (e.g. newton (power, polar))?

At my own fork of the matpower repository I created a pull request with a concept/idea on how to implement a power flow with a distributed slack bus. I would appreciate it if you have a quick look and give me some feedback.

rdzman, 2022-01-14:
*For reference, richardkue/matpower#1 is the PR referred to above.*

Thanks for sketching this out. I'm a bit torn about how to respond. The approach looks fine if you need the capability right away, but honestly, I'm hesitant about promising it'll be included in a release.

And it's not because there is anything wrong with it. I'm definitely interested in including distributed slack in MATPOWER's power flow, but I'm also well into what amounts to a massive MATPOWER re-design/re-write that will deprecate this code with the next release. My preference would be for you to work on implementing distributed slack in the new code (see [MP-Element](https://github.com/MATPOWER/mp-element)). It should be cleaner to implement there, but the downside is that the code is more of a moving target, is not well documented yet, and is still incomplete in certain areas.

I'm happy to schedule a call to discuss if that would be helpful. We can coordinate by e-mail if interested.

richardkue, 2022-01-18:
Thanks for your feedback. I understand your concerns about the design. I also see it as a bit of a code degradation.
I will have a look at MP-Elements. But I am not yet familiar with it. I will come back in the next weeks with a new idea.

SharkpJIM, 2025-05-21:
Cảm ơn Giáo sư! hiện nay tôi đang nghiên cứu về Vận hành tối ưu trên lưới phân phối có xét đến DG của EVS, nhưng code funtion của Matlab là chủ yếu. Trong đó, phân tích dòng điện AC và DC khi sạc là chính, tôi muốn tham khảo từ các bản code của Giáo sư riêng về phần tối ưu công suất trên lưới, và trong nghiên cứu sắp tới.