MATPOWER/matpower #137: Usage of `Bs` value in bus table

claytonpbarrows, 2022-02-10:
I'm comparing the results from `run_pf` of MATPOWER and [PowerModels.jl](https://github.com/lanl-ansi/powermodels.jl) and [PowerSystems.jl](https://github.com/nrel-siip/powersystems.jl). I'm finding discrepancies when there are nonzero `Bs` values defined in the bus table of a MATPOWER case. It appears that these are not used in the MATPOWER `run_pf` routine. Is there a reason for omitting the representation of the shunts from the power flow? 
The comparison is being conducted here: https://github.com/G-PST/power-flow-exercise/tree/cb/fix_psy_pf/example-powersystems

rdzman, 2022-02-11:
... and line 0091.

What is the difference you are seeing? Is the voltage solution different when you have non-zero `Bs` values? If so, the first place I would look is to make sure the shunt values are being converted to p.u. in the Julia packages. In the MATPOWER case format they are specified in MVAr injected at 1 p.u. voltage, so they must be divided by `baseMVA` to get the p.u. admittance values.

claytonpbarrows, 2022-02-11:
Thanks for the responses. I now understand a couple details that explain the differences that I've been encountering.
1. The bus Qd values reported by MATPOWER do not include the shunt injections
2. The branch Q injections reported in the MATPOWER summary do not include shunt injections, but the branch Loss calculation does

@abhyshr  and @rdzman, if I'm wrong on these points, please correct me.

closed 2022-02-11 by claytonpbarrows

abhyshr, 2022-02-14:
1. Yes
2. Ray can correct me on this one. The branch losses and reactive injections only report the contributions from the branches, not the bus shunts. The branch loss is the series loss and the reactive Q injection comes from the line charging suspectance `Bc`.  The bus shunt injections are only reported as an aggregate, not on per-bus basis.

rdzman, 2022-02-14:
1. Correct.
2. @abhyshr is correct here as well.

In the `System Summary` section ...
- the bus shunt elements are only included in `Shunts (inj)`, not in `Branch Charging (inj)`
- `Branch Charging (inj)` includes _only_ the branch shunt elements representing the line charging susceptance `Bc`
- `Losses (I^2 * Z)` is _only_ the series losses (not including any bus or branch shunt components)