MATPOWER/matpower #138: A question about `get_losses`

christian-cahig, 2022-02-22:
The example in the help string reads

https://github.com/MATPOWER/matpower/blob/46def0f500d63c83916ae907c2e5cb22fd7207c1/lib/get_losses.m#L48-L50

I was wondering if calculation of the total system reactive losses can be done in a similar manner, *i.e.*,

```matlab
results = runpf(mycase);
total_system_reac_losses = sum(imag(get_losses(results)))
```

or are there some extra steps necessary?

rdzman, 2022-02-22:
That gives you the total reactive losses from the series portion of the branch model, but those are offset by reactive injections from the line charging elements of the branch model. So for the system total of the net reactive line losses you would subtract the charging injections from the series losses.
```
results = runpf(mycase);
[loss, chg] = get_losses(results);
total_system_real_losses = sum(real(loss));
total_system_reac_losses = sum(imag(loss)) - sum(ctg);
```

rdzman, 2022-02-22:
That is correct.

christian-cahig, 2022-02-23:
@rdzman, I think it would be nice if your example above, *i.e.*

> ```
> results = runpf(mycase);
> [loss, chg] = get_losses(results);
> total_system_real_losses = sum(real(loss));
> total_system_reac_losses = sum(imag(loss)) - sum(chg);
> ```

is added to the example in the help text. What do you think? I can work up a quick PR for it.

rdzman, 2022-02-23:
Good idea. Thanks for the PR!

christian-cahig, 2022-02-24:
@rdzman you're welcome. I'll close this issue now.

closed 2022-02-24 by christian-cahig