MATPOWER/matpower #140: OPF error

bmitrauncc, 2022-02-25:
I wanted to run a simple OPF but receiving this error:

Warning: The following error was caught while executing 'osqp' class destructor:
Undefined function 'osqp_mex' for input arguments of type 'char'.

Error in osqp/delete (line 40)
            osqp_mex('delete', this.objectHandle);

Error in osqp (line 34)
            this.objectHandle = osqp_mex('new', varargin{:});

Error in have_feature_osqp (line 23)
        o = osqp();

Error in have_feature (line 135)
            [TorF, vstr, rdate] = feval(fcn);

Error in mpoption_info_osqp (line 31)
if have_feature('osqp')

Error in mpoption>mpoption_default (line 1638)
            opt = nested_struct_copy(opt, feval(fname, 'D'));

Error in mpoption (line 643)
    opt = mpoption_default();

Error in runopf (line 66)
            mpopt = mpoption;       %% use default options

Error in running_opf (line 2)
results = runopf(mp); 
> In osqp (line 34)
In have_feature_osqp (line 23)
In have_feature (line 135)
In mpoption_info_osqp (line 31)
In mpoption>mpoption_default (line 1638)
In mpoption (line 643)
In runopf (line 66)
In running_opf (line 2) 
OSQP Error!

rdzman, 2022-02-25:
Hmmm. Is it possible you have a partial install of OSQP in your Matlab path?

What is the output of the following commands (in a fresh MATLAB session)?
```
which osqp
which osqp_mex
mpver
path
```

bmitrauncc, 2022-02-25:
Hi please see the outputs

>> which osqp
C:\Users\mitr284\Work\Software\matpower7.1\matpower7.1\mp-opt-model\.github\osqp\osqp.m  % osqp constructor
>> which osqp_mex
'osqp_mex' not found.
>> mpver

MATPOWER               Version 7.1        08-Oct-2020
MATLAB                 Version 9.9        29-Jul-2020   Release: (R2020b)  
Optimization Toolbox   Version 9.0        29-Jul-2020   Release: (R2020b)  
MP-Test                Version 7.1        08-Oct-2020
MIPS                   Version 1.4        08-Oct-2020
MP-Opt-Model           Version 3.0        08-Oct-2020
MOST                   Version 1.1        08-Oct-2020
SDP_PF                 Version 1.0.2      08-Oct-2020
SynGrid                Version 1.0.1      20-Jun-2019
Artelys Knitro         -- not installed --
BPMPD_MEX              -- not installed --
CLP                    -- not installed --
CPLEX                  -- not installed --
GLPK                   -- not installed --
Gurobi                 -- not installed --
IPOPT                  -- not installed --
MINOPF                 -- not installed --
MOSEK                  -- not installed --
Warning: The following error was caught while executing 'osqp' class destructor:
Undefined function 'osqp_mex' for input arguments of type 'char'.

Error in osqp/delete (line 40)
            osqp_mex('delete', this.objectHandle);

Error in osqp (line 34)
            this.objectHandle = osqp_mex('new', varargin{:});

Error in have_feature_osqp (line 23)
        o = osqp();

Error in have_feature (line 135)
            [TorF, vstr, rdate] = feval(fcn);

Error in osqpver (line 19)
o = have_feature('osqp', 'all');

Error in mpver (line 189)
    osqpver; 
> In osqp (line 34)
In have_feature_osqp (line 23)
In have_feature (line 135)
In osqpver (line 19)
In mpver (line 189) 
OSQP Error!
OSQP                   -- not installed --
PARDISO                -- not installed --
PDIPMOPF               -- not installed --
SCPDIPMOPF             -- not installed --
SDPT3                  -- not installed --
SeDuMi                 -- not installed --
TRALMOPF               -- not installed --
YALMIP                 Version 20210331   31-Mar-2021
Architecture:          PCWIN64

  MATPOWER 7.1 is distributed under the 3-clause BSD License.
  Please see the LICENSE file for details.

>> path

		MATLABPATH

	C:\Users\mitr284\OneDrive - PNNL\Documents\MATLAB
[... 1030 lines cut ...]
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\toolbox_packaging
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\toolstrip
	C:\Program Files\MATLAB\R2020b\toolbox\shared\toolstrip_coder_app\plugin\m
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\uicomponents\uicomponents\plugin\appdesigner
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\uicomponents\uicomponents
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\uicomponents\uicomponents\graphics
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\uicomponents\uicomponents\style
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\uicomponents\uicomponents\databrowser
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\uicomponents\uicomponents\uicontrol
	C:\Program Files\MATLAB\R2020b\toolbox\shared\spcuilib\unifiedscopes
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\updateinstaller
	C:\Program Files\MATLAB\R2020b\toolbox\shared\viewmodel\viewmodel
	C:\Program Files\MATLAB\R2020b\toolbox\simulink\vmgr\datamodel\web\UI
	C:\Program Files\MATLAB\R2020b\toolbox\simulink\vmgr\plugin\m
	C:\Program Files\MATLAB\R2020b\toolbox\wavelet\wavelet
	C:\Program Files\MATLAB\R2020b\toolbox\wavelet\wmultisig1d
	C:\Program Files\MATLAB\R2020b\toolbox\wavelet\compression
	C:\Program Files\MATLAB\R2020b\toolbox\wavelet\wavedemo
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\webcam
	C:\Program Files\MATLAB\R2020b\toolbox\shared\spcuilib\webscopes
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\external\interfaces\webservices\http
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\external\interfaces\json
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\external\interfaces\webservices\restful
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\external\interfaces\webservices
	C:\Program Files\MATLAB\R2020b\toolbox\matlab\external\interfaces\webservices\wsdl

rdzman, 2022-02-25:
Ok, it's because you have lots of directories in your path that do not belong there. It looks like you just added the `matpower7.1` directory and all of its subdirectories to your path rather than using the `install_matpower` script.

Try this. Remove `C:\Users\mitr284\Work\Software\matpower7.1` and all subdirectories from your path. Then, `cd` to `C:\Users\mitr284\Work\Software\matpower7.1\matpower7.1` and run `install_matpower` to set up the path properly for MATPOWER.

Hopefully, that'll get you up and running.

bmitrauncc, 2022-02-25:
Thank yo so much it worked perfectly

closed 2022-02-25 by bmitrauncc