MATPOWER/matpower #142: Fail to detect Knitro 13.1

Gwzwpxz, 2022-07-27:
Hello,

I'm trying to use MATPOWER v7.1 with Knitro 13.1 on Linux but it cannot detect Knitro properly.
I noticed that Knitro now uses "knitro_matlab" instead of "knitromatlab" which appears in "have_feature_knitromatlab()" and
the inconsistency appears in several Knitro API functions (knitro_milp, knitro_nlp, etc.).

Best regards

rdzman, 2022-07-27:
Thanks for reporting this. I had already fixed this issue in [MP-Opt-Model](https://github.com/MATPOWER/mp-opt-model), but have not yet merged it into the `master` branch or pulled it back into MATPOWER. I'll try to do that now so you can simply pull the latest MATPOWER `master` to get it working again.

rdzman, 2022-07-27:
Ok, that fix has been merged into `master` now, so cloning or downloading from this repo should fix the problem.

closed 2022-07-27 by rdzman

Gwzwpxz, 2022-07-28:
Thanks for your prompt reply! It now works fine.