MATPOWER/matpower #143: Faster flow derivatives

SanPen, 2022-07-29:
Hi, 

I am not entirely sure this belongs here since it is not a bug or a software issue per se.

Following matpower's technical notes, I've implemented the branch flow derivatives. These expressions require several sparse matrix multiplications, and additions. I've used them in the past and they're great. However, when I look the same derivatives in the literature (the available state estimation books [1](https://www.amazon.com/State-Estimation-Electric-Power-Systems/dp/0792385195) and [2](https://www.amazon.com/Power-System-State-Estimation-Implementation/dp/0824755707/ref=sr_1_1?crid=31JQFSRB9PC2I&keywords=state+estimation+ali+abur&qid=1659092816&s=books&sprefix=state+estimation+ali+abur%2Cstripbooks-intl-ship%2C164&sr=1-1)) I find that the provided branch flow expressions do not contain any summation and are quite straight forward.

This lead me to research a bit deeper, and implement Matpower expressions, Numerical differentiation, Monticelli's Expressions and Gomez Expósito's expressions [here](https://github.com/SanPen/GridCal/blob/master/src/research/derivatives_and_jacobian/branch_power_2.py).

The outcome is that the derivatives with respect to the voltage module present a significant deviation, whereas the derivatives w.r.t the voltage angle are exactly the same when comparing the Matpower derivatives with the ones provided in those books.

Honestly, I trust better the derivatives from Matpower, but the book's expressions provide a massive computational gain.

I'd appreciate any insight on this.

Best regards,
Santiago

rdzman, 2022-08-02:
Thanks @SanPen for sharing this, and yes, it is entirely appropriate here. I'm always open to improvements to MATPOWER performance, especially those that are simple and have broad impact, such as computation of power flow derivatives.

So, if I understand correctly, you are saying that the expressions from Monticelli and Expósito both result in identical numerical values to those produced by MATPOWER, but with a significant savings in computation time, correct?

If so, please consider submitting a pull request with updates to MATPOWER's routines (or simply sending me an updated version of the appropriate MATPOWER function). It looks like your comparison code is written in Python, which may have different performance characteristics, so I would need to see a comparison written in MATLAB and tested across a wide range of network sizes.

It would be great if we can get an easy and significant speedup.

SanPen, 2022-08-02:
Hi, 

In the past, by exploiting the sparse structure of Ybus represented in column major (CSC), it was possible to speedup the jacobian compared to the matrix operations about an order of magnitude. This in a python implementation where the optimized function was written using a just-in-time compiler (the numba package). See [this](https://github.com/SanPen/GridCal/blob/master/src/GridCal/Engine/Simulations/PowerFlow/NumericalMethods/derivatives.py) for the implementations.

In matlab, I've read that such JIT happens anytime a loop is detected and without having to do anything other than not vectorizing the operations. I'm quite rusty in Matlab, but it seems that you cannot access the internal sparse matrices storage directly, but using the `find` function that provides the COO representation. For the flow derivatives the best would be a CSR representation.

The derivatives I've found come from state estimation algorithms, and it seems that they provide the flow derivative seen from the "from" bus and probably not taking into account the taps that may be present in a transformer... So I'll need some more research, but my goal would be the _about and order of magnitude_ speedup.

I'll keep you updated, and eventually I'll provide some matlab functions.

rdzman, 2022-08-02:
Thanks. That would be wonderful.

SanPen, 2022-08-03:
Hi, I decided to derive my own expressions and it turned out a good investment.

I attach a file with the function to compute the derivatives, a file to call it and compare it and a text file with the derivatives. It is not faster than the matlab matrix implementation but for now I blame it on not having the internal structure right away and having to use the `find` function.

See the attached file [flow_derivatives.zip](https://github.com/MATPOWER/matpower/files/9250734/flow_derivatives.zip)

TBD...

rdzman, 2022-08-03:
Very interesting. I modified your comparison code to run through a number of large cases I have and the speedup ranges from about 1.3x to 2.5x with an average of about 1.8x. Also, in the one case I profiled the `find` lines only accounted for about 3% of the time spent in the functions, so pretty negligible.

On the other hand, this is an incredible example of the value of a JIT compiler, which MATLAB has and Octave does not. When I ran the same tests on Octave, it resulted in about a 50x slow-down!

I haven't looked at the details yet, but I'm wondering if there is a way to vectorize your code and get rid of the loops. If so, it's likely Octave could also see a speed-up.

cuihantao, 2022-10-19:
This is an interesting discussion! @rdzman, by the way, MATLAB does have a JIT compiler introduced several years ago. It is transparent to the user. See https://www.mathworks.com/products/matlab/matlab-execution-engine.html

rdzman, 2022-10-19:
> This is an interesting discussion! @rdzman, by the way, MATLAB does have a JIT compiler introduced several years ago. It is transparent to the user. See https://www.mathworks.com/products/matlab/matlab-execution-engine.html

Exactly. My point was that MATLAB does, and Octave does not ... resulting in speedups for MATLAB and massive slow-downs for Octave with the proposed changes.

cuihantao, 2022-10-19:
> Exactly. My point was that MATLAB does, and Octave does not ... resulting in speedups for MATLAB and massive slow-downs for Octave with the proposed changes.

Thanks for your reply. I got it now. Sorry that I misread your original post :)

cuihantao, 2022-10-20:
I thought about this again and recalled that PSAT implements a rather compact form for the Jacobians. The derivatives `d S/d V` and `d S/d \theta` are calculated in two equations, then the real and imaginary parts are taken to find the four blocks in the Jacobian. See https://github.com/Sinan81/PSAT/blob/master/psat-mat/psat/%40LNclass/build_gy.m.

Note that in his code, the variable `dR = j * conj( d S/ d \theta)`, so that the real and imaginary parts are swapped. The variable named `dS` in the code is for `d S/ d V`.

The expressions heavily uses matrix mul by sparse diagonal matrices, thus it can be quite efficient. This might be worth benchmarking as well.

SanPen, 2022-10-20:
Hi!
The fastest that seems possible is [this](https://github.com/SanPen/GridCal/blob/aaa4ad6cec0c637f67194e98cf0b615f16b6711b/src/GridCal/Engine/Simulations/PowerFlow/NumericalMethods/derivatives.py#L114). It comes from a 2-pass algorithm that takes into account the CSC or CSR structure of the matrix. 
The trick here is that the derivatives `dS/dV` have the same sparse structure as `Ybus`.

cuihantao, 2022-10-20:
Hi!

Thanks for your reply. From https://github.com/Sinan81/PSAT/blob/b0519c6cf5e87c6cbc5360bcc973a3cfdddcd864/psat-mat/psat/%40LNclass/build_gy.m#L18-L19, I see that the derivatives `dS/dV` and `dS/d\theta` _may_ both have the same structure as Ybus, because multiplying `Ybus` by a full-rank sparse diagonal matrix will result in a matrix with the same shape as `Ybus`. Some locations may end up with zero values though, but they can be kept in the compressed sparse matrix.

ANDES populates an allocated, zero-filled Jacobian sparsity pattern matrix before calculating the Jacobians. To inspect the sparsity pattern matrix, you can do 

```python
import andes
ss = andes.system.example()  # 14-bus system

ss.dae.gy
```
The first 28-by-28 matrix is the Jacobian for network equations. Inspect the patterns of `dP / d\theta` and `dP / dV`:

```python
print(ss.dae.gy[:14, :14])
print(ss.dae.gy[:14, 14:28])

len(ss.dae.gy[:14, :14]) == len(ss.dae.gy[:14, 14:28])  # same number of possible fills
```
 
Again, the allocated entries in the pattern matrix may be updated to non-zeros or not. In general, updating a preallocated sparse matrix is a lot faster. Not sure how much this can be used for MATPOWER or GridCal though.

rdzman, 2022-10-20:
> The expressions heavily use matrix mul by sparse diagonal matrices, thus it can be quite efficient. This might be worth benchmarking as well.

This sounds very similar to what MATPOWER already does. Is it different?

SanPen, 2022-10-20:
At least for python, the matrix multiplication approach is an order of magnitude slower than the 2-pass algorithm. It is true that scypi's sparse module is notoriously inefficient. For the C++ software we do at REE, the matrix version was not terrible but the 2-pass algorithm is unbeatable to my knowledge. For matlab, since there is no way to compose a CSC matrix directly from the pointers it is hard to tell.

cuihantao, 2022-10-21:
> This sounds very similar to what MATPOWER already does. Is it different?

I quickly checked your MATPOWER code, and yes, they are indeed quite similar. Using matrix product, the implementation in MATPOWER is already quite efficient.

cuihantao, 2022-10-21:
> At least for python, the matrix multiplication approach is an order of magnitude slower than the 2-pass algorithm. It is true that scypi's sparse module is notoriously inefficient. For the C++ [...]

Your analysis makes sense to me. Have you tested the sparse matrix implementation in CVXOPT/KVXOPT? I vaguely remember that it scales differently than SciPy's.

SanPen, 2022-10-21:
It was a big disappointment when the code metrics showed that scipy's sparse operations were the bottleneck. So I went the C++ route altogether. I've heard that CVXOPT/KVXOPT is faster. However if I ever change GridCal's matrices it'll be by armadillo's python wrapper, since Armadillo is flexible enough and it is properly designed.

closed 2023-02-28 by rdzman