MATPOWER/matpower #144: warning: matrix singular to machine precision

yanda1234, 2022-08-10:
Hi,

While using matpower to analyze one system created by myself, I encountered a problem, which pushed me here after struggling with  it for quite a while. 
This is the result I got:
[image]
This is the case information I got:
[image]
[image]
[image]

I have checked that total system capacity should be enough to meet the load and there is no island existed.
Besides, I also tried to increase branches' capacities over 100 times. However, the problem is still not solved.
Could you please shed me some lights on where the problem might be?
Many thanks.

Best,
Yanda

rdzman, 2022-08-11:
Is it possible that some branches have very small (or zero) reactance? That would cause singular matrix issues.

rdzman, 2022-08-15:
> Thanks for your reply. There were some zero reactance branches there. **However, after I set smallest positive reactance value to be 1e-3 and increase all branches' rateA capacities to be 100 times, the problem is still there**. Do you know any other possible reasons?

First of all, the `rateA` capacity is not used at all for a simple power flow. It's only for OPF.

So, is it still the same singular matrix problem, or is it simply non-convergence now? The singular matrix issue is normally related to zero impedance branches or networks with islands.

> Is there any upper limit for system size which matpower can handle? Btw, I notice psse can handle system with negative or zero impedance branches. Do you know reason for that?

No, there is no explicit size limit on the systems MATPOWER can handle.

I assume that internally PSS/E does transformations to eliminate zero impedance branches by combining the connected nodes, then splitting them out again for the result reporting. MATPOWER does not yet automate that process.
 
> **In addition, I also used the below way to check system load, but the continuation power flow also diverges. The most weird thing is even after I:
> 
> 1. removed all of system active and reactive load and then set generators' Pg, Qg to be zero;
> 2. set generators' Qmax to be 1e6 and Qmin to be -1e6.
>    it still cannot converge. How could this happen?**
>    (All generators' bus type have been set to type 2, except one reference bus. And all system buses are consecutive.)

The CPF needs to be able to solve an initial power flow and it looks like that is what is failing. Maybe the initial bus voltages are not close enough to the solution. Have you tried with a flat start? Are there phase shifting transformers in the system, and if so, do the starting voltage angles match the phase shifts?

Also, FYI, by default, the power flow also does not take into account the generator reactive limits. It simply solves the power flow given the specified load and generation.

rdzman, 2022-08-18:
_(For future reference, it's best to start a new issue or mailing list post for a new question)._

Regarding MATPOWER's handling of PSS/E RAW files, version 33 was the latest version available last time I worked on the `psse2mpc()` code. So, I'm not sure how 34 and 35 are different and how that would affect the import routines. There is a chance it could still work if the data utilized by MATPOWER in your particular case is not affected. But the bottom line is that it's completely untested with 34 and 35.

yanda1234, 2022-08-18:
Yes, I spent several days trying to update matpower code, but then I realize there is a easy way, which is use psse to save case as previous version. Thanks, Dr. Zimmerman!

yanda1234, 2022-08-24:
Hi Dr. Zimmerman,

I found this part in psse_convert_xfmr.m file. It seems matpower regards reactance of delta-connected three-winding transformer is actually impedance. I am curious of the reason.
[image]
Besides, I noticed matpower treats two-term DC lines as pipes of power flow and didn't keep rest parameters, including Rdc, Rcmp_ohm, Delti, Vschedule etc. Could you please shed me with some light on your reason?
Thanks a lot in advance.

Best,
Yanda

rdzman, 2022-08-24:
> I found this part in psse_convert_xfmr.m file ...

Please create a new issue for this question, and close this one unless the singular matrix issue has not yet been resolved.

yanda1234, 2022-08-24:
> > I found this part in psse_convert_xfmr.m file ...
> 
> Please create a new issue for this question, and close this one unless the singular matrix issue has not yet been resolved.

Hi Dr. Zimmerman,

I am working on these two things is just to solve the in-convergence issue. I have identified the in-convergence was caused by 2-term DC line or delta-star transfer of three-winding xfmr and thus I turned to them.

Best,
Yanda

rdzman, 2022-08-24:
The PSS/E import code was originally written by Yujia Zhu, so I'm not sure of all of the details, but I believe the handling of 3-winding transformers was based on [this article](https://www.yumpu.com/en/document/view/10534310/modeling-of-three-winding-voltage-regulating-transformers), which may answer your question.

The reason for the DC modeling is simply that it was easy to implement as a pair of linked injections (fake generators) and that was a decent 1st order approximation to the behavior of a DC line. Implementing a proper detailed DC line model would have taken significantly more effort.

However, the latest development MATPOWER with [MP-Element](https://github.com/MATPOWER/mp-element/) makes it possible to implement much more sophisticated element models easily, so hopefully we can include a proper DC line model in the future. Contributions are always welcome.

closed 2022-10-26 by yanda1234