MATPOWER/matpower #146: MATPOWER FOR POWER FLOW ANALYSIS OF ISLAND MICROGRID WITH VIRTUAL SYNCHRONOUS MACHINES (VISMA)

sanusielect, 2022-08-21:
I am a research student working on multi machine analysis of Virtual synchronous machine (n-VISMA) with constant impedance loads. In the present stage of my work, i need to carry out steady state load flow analysis to obtain the voltage at each bus, the steady stage generators output current and active power P and reactive power Q injected at each buses.  

In conventional load flow analysis, single generator bus is typically chosen as a reference bus to serve the roles of both a
voltage angle reference and a real power slack while other generator buses are treated as PV buses with voltage magnitude and
generator real power injection given.

In the my case, the voltage magnitude and active power injection at the buses are not known but rather they are to be determined. All  the constant impedance loads have been eliminated by kron reduction so that direct interactions among the generators can be studied.

Please how can go I about this? Please can you refer me to any relevant MATPOWER cases that can be of help? I am considering an unregulated machines now with no droop buses.

Below is the simple schematic of the model for two machine cases with a constant impedance load

[image]

Sir, your guidance would be so much appreciated

Thank you and kind regards

rdzman, 2022-08-22:
What quantities are known? As it is, It sounds to me like your problem has an infinite number of solutions, including zero voltage and zero flows.

In order to solve a power flow, you need to specify half of the nodal variables (P, Q, V, theta) and solve for the other half.

sanusielect, 2022-08-24:
Thank you for input professional input Prof. The mechanical virtual torques are known (Tm1 & Tm2), Internal generator node peak voltage (E1 & E2) are specified but the angles are not known, all the line parameters are known. If the machine is to be regulated by droop control then generator terminal buses 1 & 2 can be considered as droop buses while the load bus (Zload) as PQ bus, then the load flow can easily be carried out but my case is considering an unregulated machine with specified constant torque to each machine. Now, the generator terminal bus voltages at nodes 1 & 2 and power injection at the buses are needed to be determined

rdzman, 2022-08-25:
I'm afraid my generator modeling is rusty, so I'm not sure about the implications of constant torque, but I suspect it implies some constraint on the relationship between the voltage and power at the generator.

If so, you should be able to solve the problem with MATPOWER using a customized OPF. You'll want to use a constant impedance load (set `exp.sys_wide_zip_loads.pw` option to `[0 0 1]`). And you'll need to add the appropriate constraint to the generator power and gen bus voltage. You should be able to do that via direct specification as described in Section 7.1.2 of the manual. If it is a linear constraint, specify `mpc.A`, `mpc.l`, `mpc.u`. If it's nonlinear, then you'll have to implement the constraint function and its derivatives and specify them using `mpc.user_constraints`.

closed 2022-11-09 by rdzman