MATPOWER/matpower #147: Derivation of HVDC Qlim

yanda1234, 2022-08-25:
Hi Dr. Zimmerman,

Is it possible to know how you derived this function, i.e "psse_convert_hvdc_Qlims"? (such as a document for explanation)
[image]
Thanks a lot.

Best,
Yanda

rdzman, 2022-08-25:
I'm afraid I don't have anything more than what is included in that comments, with a reference to Kimbark's book. I'm assuming these comments were written by the original author, Yujia Zhu, who was a student at Arizona State University at the time.

yanda1234, 2022-08-25:
> I'm afraid I don't have anything more than what is included in that comments, with a reference to Kimbark's book. I'm assuming these comments were written by the original author, Yujia Zhu, who was a student at Arizona State University at the time.

I see. Thanks, Dr. Zimmerman.

closed 2022-08-25 by yanda1234