MATPOWER/matpower #148: delta-star transformation

yanda1234, 2022-08-29:
Hi Dr.Zimmerman,

I just notice matpower implements delta-star transformation in this way. 
[image]
The correct way should be as below, right?
[image]

Best,
Yanda

rdzman, 2022-08-30:
It looks to me like this code is based on Eqs. 2 from [this article](https://www.yumpu.com/en/document/view/10534310/modeling-of-three-winding-voltage-regulating-transformers) and is correct.

However, if you still have doubts after looking at that, my only suggestion would be to attempt to contact the original author of the code, Yujia Zhu, to get more insight or confirmation. Let me know if you feel that's necessary and I'll see what I can do.

yanda1234, 2022-08-30:
I see your point. Thanks for your clarification, Dr. Zimmerman!

closed 2022-08-30 by yanda1234