MATPOWER/matpower #153: DC power flow converge, but DCOPF diverge

yanda1234, 2022-11-08:
Hi,

I encountered a weird issue. The DC power flow can converge, but the DC OPF cannot, even after all branches' rate_A have been set as 0 (i.e no limit). How could this happen?
[image]
[image]

Kindly advise. Many thanks!

Best,
Yanda

rdzman, 2022-11-09:
First of all, which solver are you using?

I'm not recognizing the cause of the warning message you are getting, but I am suspicious about whether the modeling is valid given the range of voltage angles coming out of the DC PF (from -20 deg to 327 deg).

I'm also not sure what is causing the difference with PSS/E. It's possible they have controls that are active by default during the power flow run that MATPOWER does not include. I'm pretty sure that's the case for AC power flow, but I don't know about DC.

yanda1234, 2022-11-09:
> First of all, which solver are you using?
>
> I'm not recognizing the cause of the warning message you are getting, but I am suspicious about whether the modeling is valid given the range of voltage [...]

Hi Dr. Zimmerman,

Thanks for your prompt reply. Could you please clarify a little bit about your concern of model's validity? What kind of system model can be judged as invalid?

Kindly advise.

Best,
Yanda

rdzman, 2022-11-11:
The DC power flow simply solves an A * x == b problem, so the default solver for that is fine if A is non-singular. I was referring to the solver for the DC OPF. And the _default_ depends on what solvers you have installed. If you are able to take advantage of their [free academic license](https://www.gurobi.com/academia/academic-program-and-licenses/), I recommend [Gurobi](https://www.gurobi.com/). MOSEK and CPLEX are also good options. If those are not available, MATLAB's `linprog` (requires Optimization Toolbox) or possibly GLPK.

In terms of the model's validity, the DC power flow model may be a reasonable approximation to the full non-linear AC model **if** the simplifying assumptions are valid (see Section 3.7 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf)). One of them is that branch angle differences are small enough that the sine of the angle can be approximated by the angle itself in radians. If the DC power flow is producing large angle differences, then it is not a very good approximation to the original AC model.

From the new warning message, it looks like there may be an issue with `gencost`. You may want to double-check to be sure you are specifying valid linear, quadratic or piecewise linear costs for each generator.

The ill-conditioning could be from impedances with very small absolute value ... or possibly (not sure) from the cost issue above.

As long as there are no branch flow limits, a solved DC power flow will be a feasible solution for the OPF if and only if all of the generators were set to feasible dispatches and the solved slack generator dispatch is also within its bounds.

yanda1234, 2022-11-12:
Thanks a lot, Dr. Zimmerman. Is there any manual to explain how to connect Gurobi with matpower? I tried many times, but didn't make it yesterday.

rdzman, 2022-11-12:
Nothing special for MATPOWER. Simply install Gurobi according to their instructions for the Matlab interface. Then restart MATLAB for MATPOWER to recognize it.

Maybe [this link](https://support.gurobi.com/hc/en-us/articles/4533938303505-How-do-I-install-Gurobi-for-Matlab-) will help.

If you're still having trouble, let me know.

yanda1234, 2022-11-13:
Thanks, Dr. Zimmerman. When we tried to solve ill-conditioned issue, do you know if there is an upper and bottom bound for impedance values? What I found is when system size is very large, like over 20000 buses, the range of impedance becomes very sensitive. Is this related to solver types?

rdzman, 2022-11-14:
Overall, I think MOSEK may be better than `linprog`, but my experience is rather limited.

I have also run into problems with GLPK on some problems and have encountered cases where it helps to turn off the scaling and or pre-solve options.

closed 2022-11-14 by yanda1234

rdzman, 2022-11-14:
I've not used MOSEK with Octave in MATPOWER, so that interface is untested. If you do get MOSEK working with Octave, please share your experience as I'd like to give it a try too. Do you know if they provide the MEX interface or do you have to build your own?

rdzman, 2022-11-15:
Yeah, it looks like that interface is about 9 or 10 years old and only works for MOSEK 6. Hopefully, you can get a Matlab license.

yanda1234, 2022-11-15:
I just got it this morning. Thx, Dr. Zimmerman!

yanda1234, 2022-11-19:
> The DC power flow simply solves an A * x == b problem, so the default solver for that is fine if A is non-singular. I was referring to the solver for the DC OPF. And the _default_ depends on what [...]

Hi Dr. Zimmerman,

Do you know if pyomo has a good solver for DCOPF?

Best,
Yanda

rdzman, 2022-11-21:
I'm afraid I'm not really familiar with pyomo, but I would guess that it has interfaces to at least some of the major solvers like Gurobi, CPLEX, MOSEK.

yanda1234, 2022-11-21:
I see. Thanks, Dr. Zimmerman. I misunderstood my colleague's meaning as pyomo has solvers by itself.