MATPOWER/matpower #156: is it possible to apply constraint for the sum of multiple branches' total flow

yanda1234, 2022-11-17:
Hi Dr. Zimmerman,

I am curious if there is an easy way to apply constraint for the sum of multiple branches' total flow, instead of just one branch's flow.

Best,
Yanda

rdzman, 2022-11-17:
For the DC OPF case, see Section 7.6.2 in the [User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf
).

yanda1234, 2022-11-17:
Thanks, Dr. Zimmerman. I noticed the t_case30_userfcns.m etc were not included in matpower 7.0. And some other interesting functions or cases were also deleted. Could you please explain a little bit why you removed them?

rdzman, 2022-11-17:
It was not removed. You can find it in `lib/t` in versions [7.0](https://github.com/MATPOWER/matpower/blob/7.0/lib/t/t_case30_userfcns.m) and [7.1](https://github.com/MATPOWER/matpower/blob/7.1/lib/t/t_case30_userfcns.m) and the current [master branch](https://github.com/MATPOWER/matpower/blob/master/lib/t/t_case30_userfcns.m).

yanda1234, 2022-11-17:
I see. The letter 't' here means "test", right?

closed 2022-11-18 by yanda1234

rdzman, 2022-11-18:
Yes, `t` is the test directory.