MATPOWER/matpower #158: can savecase save field interface?

yanda1234, 2022-11-20:
Hi Dr. Zimmerman,

I am trying to use function "savecase" to save a case, but the field interface seems cannot be saved.
Is it because this field is still under development?
Besides, I noticed there are some test functions like "t_makePTDF", "t_qps_matpower" etc. Could you please tell me what their difference from the original ones are?

Many thanks in advance.

Best,
Yanda

rdzman, 2022-11-21:
I'm not sure I understand your questions.

What do you mean by "field interface"? If you are referring to the interface flow limits described in Section 7.6.2 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf), then yes, `savecase` does handle that field if the interface callbacks have been enabled with `toggle_iflims(mpc, 'on')`.

These test functions are the ones that are run when you type `test_matpower`. For example, `t_makePDTF` tests the `makePTDF()` function to ensure it is functioning as expected.

yanda1234, 2022-11-21:
Nice. Thanks, Dr. Zimmerman!

closed 2022-11-21 by yanda1234