MATPOWER/matpower #160: VOLL

yanda1234, 2022-12-08:
Hi Dr. Zimmerman,

I used "load2disp" (VOLL) to convert fixed load of one power flow case into dispatchable load.
However, after running DCOPF, I am curious if I can see how VOLL changes result from somewhere.
Is there a field or variable designed for this? Kindly let me know your thoughts.
Many thanks in advance!

Best,
Yanda

rdzman, 2022-12-08:
If you set the `out.lim.pg` option to 2, the pretty printed output will show the dispatches for all of the dispatchable loads, along with their limits. These dispatches appear as negative injections.

You can also access them directly from the `gen` matrix in the results with something like the following:

```
mpopt = mpoption('out.lim.pg', 2);
i  = find(isload(mpc.gen));
r = rundcopf(mpc, mpopt);
Pd = -r.gen(i, PG)   % dispatched load
Pd_nominal = -r.gen(i, PMIN)  % nominal load
```

yanda1234, 2022-12-13:
Thanks, Dr. Zimmerman.

yanda1234, 2023-06-29:
Hi Dr. Zimmerman,

If running **mpoption('gurobi.method',1)** gives me:
**error using mpoption
mpoption: 'gurobi' is not a valid option name.**
Does that mean gurobi solver is not existed?
Kindly advise.

Best,
Yanda

rdzman, 2023-06-29:
Please open a new issue, rather than re-opening an existing unrelated issue that has been resolved.

closed 2023-06-29 by rdzman