MATPOWER/matpower #161: Mat power not converging help!

fanbigg, 2023-01-20:
mpc.version = '2';

%%-----  Power Flow Data  -----%%
%% system MVA base
mpc.baseMVA = 100;

%% bus data
%	bus_i	type	Pd	Qd	Gs	Bs	area	Vm	Va	baseKV	zone	Vmax	Vmin
mpc.bus = [
	1	1	0	0	0	0	1	1.0235	-15.82	353.107	1	1.0235	1.0235
	3	1	12.3	5	0	0	1	1.00165	-20.11	138.227	1	1.00165	1.00165
	5	1	15.3	3.2	0	0	1	1.00412	-23.95	69.285	1	1.00412	1.00412
	10	1	16.8	2.5	0	0	1	1.02183	-20.13	70.506	1	1.02183	1.02183
	12	1	22.9	6.5	0	0	1	1.0202	-21.45	70.394	1	1.0202	1.0202
	13	1	21.2	6.8	0	0	1	1.0003	-21.97	69.021	1	1.0003	1.0003
	14	1	22.2	15.2	0	0	1	1.00317	-23.8	69.219	1	1.00317	1.00317
	15	1	58.2	36.3	0	0	1	1.00614	-22.45	69.424	1	1.00614	1.00614
	16	1	57.8	40.4	0	0	1	1.00281	-22.62	69.194	1	1.00281	1.00281
	17	1	32.8	12.9	0	0	1	1.01221	-22.18	69.842	1	1.01221	1.01221
	18	1	51.2	14.8	0	0	1	0.99716	-24.72	68.804	1	0.99716	0.99716
	19	1	18.3	5	0	0	1	1.01248	-21.81	69.861	1	1.01248	1.01248
	20	1	15.3	5	0	0	1	1.01128	-22.88	69.778	1	1.01128	1.01128
	21	1	37.2	13.4	0	0	1	1.01361	-23.09	69.939	1	1.01361	1.01361
	24	1	36.3	10.4	0	0	1	1.00265	-22.87	69.183	1	1.00265	1.00265
	27	1	19.5	7.8	0	0	1	1.00483	-22.58	69.333	1	1.00483	1.00483
	28	2	0	0	0	0	1	1.03	-12.79	355.35	1	1.03	1.03
	29	1	0	0	0	0	1	1.02446	-15.39	141.376	1	1.02446	1.02446
	30	1	23.4	6.2	0	0	1	1.00822	-19.01	139.134	1	1.00822	1.00822
	31	3	0	0	0	0	1	1.03	-14.78	355.35	1	1.03	1.03
	32	1	0	0	0	0	1	1.01126	-18.5	139.553	1	1.01126	1.01126
	33	1	14	3	0	0	1	1.01618	-19.68	70.116	1	1.01618	1.01618
	34	1	44.62	0	0	0	1	1.00169	-24.09	69.117	1	1.00169	1.00169
	35	1	0	0	0	0	1	1.02487	-16.09	141.432	1	1.02487	1.02487
	37	1	33.4	9.6	0	0	1	0.99525	-24.76	68.672	1	0.99525	0.99525
	38	1	0	0	0	0	1	1.02471	-15.75	353.525	1	1.02471	1.02471
	39	1	0	0	0	0	1	1.01439	-18.08	139.986	1	1.01439	1.01439
	40	1	0	0	0	0	1	1.00284	-19.82	138.392	1	1.00284	1.00284
	41	1	0	0	0	0	1	1.00265	-20.16	138.366	1	1.00265	1.00265
	44	2	59.8	12.3	0	0	1	1.02	-22.48	70.38	1	1.02	1.02
	47	1	0	0	0	0	1	0.99818	-20.48	137.749	1	0.99818	0.99818
	48	1	55.8	12.5	0	0	1	1.01963	-22.71	70.354	1	1.01963	1.01963
	50	2	14.1	2	0	0	1	1.02	-20.82	70.38	1	1.02	1.02
	53	1	59.5	27.8	0	0	1	1.00146	-20.47	138.201	1	1.00146	1.00146
	54	2	12.43	5.73	0	0	1	1.01	-22.35	69.69	1	1.01	1.01
	55	1	45.3	12.3	0	0	1	0.99019	-23.13	68.323	1	0.99019	0.99019
	56	1	14	3.7	0	0	1	1.02344	-15.91	141.235	1	1.02344	1.02344
];

%% generator data
%	bus	Pg	Qg	Qmax	Qmin	Vg	mBase	status	Pmax	Pmin	Pc1	Pc2	Qc1min	Qc1max	Qc2min	Qc2max	ramp_agc	ramp_10	ramp_30	ramp_q	apf
mpc.gen = [
	14	0	0	5	0	1.02	100	0	35	10	0	0	0	0	1;
    28	150	-0.45	60	-60	1.03	100	1	150	0	0	0	0	0	10;
    28	150	-0.45	60	-60	1.03	100	1	150	0	0	0	0	0	10;
    31	217.64	54	120	-90	1.03	100	1	220	0	0	0	0	0	10;
    44	20	29.68	40	-20	1.02	100	1	150	20	0	0	0	0	1;
    48	0	0	26	-14	1	100	0	52	16	0	0	0	0	1;
    50	42.09	1.82	33	-18	1.02	100	1	80	38	0	0	0	0	1;
    53	140	45	45	0	1.02	100	1	400	0	0	0	0	0	1;
    54	106.08	7.68	60	-20	1.01	100	1	110	15	0	0	0	0	1;
];

%% branch data
%	fbus	tbus	r	x	b	rateA	rateB	rateC	ratio	angle	status	angmin	angmax
mpc.branch = [
	1	31	0.0017	0.017	0.182	597	597	597	0	0	1	-360	360;
	1	40	0.001	0.0623	0	250	250	250	2.5	0	1	-360	360;
	3	40	0.0048	0.0368	0.0182	233	233	233	0	0	1	-360	360;
	3	41	0.00902	0.0571	0.0135	233	233	233	0	0	1	-360	360;
	5	44	0.03463	0.08253	0.0016	82	82	82	0	0	1	-360	360;
	5	18	0.03133	0.07675	0.0015	82	82	82	0	0	1	-360	360;
	10	13	0.03375	0.0789	0.0012	82	82	82	0	0	1	-360	360;
	10	19	0.0405	0.0953	0.0021	72	72	72	0	0	1	-360	360;
	10	39	0.00106	0.03949	-0.0041	186.7	186.7	186.7	0.5	0	1	-360	360;
	12	27	0.0142	0.07557	0.0695	112	112	112	0	0	1	-360	360;
	12	18	0.02747	0.08909	0.0019	106	106	106	0	0	1	-360	360;
	12	17	0.02172	0.06935	0.0019	82	82	82	0	0	1	-360	360;
	12	40	0.00107	0.04039	-0.0042	187	187	187	0.5	0	1	-360	360;
	12	40	0.00107	0.04039	-0.0042	187	187	187	0.5	0	1	-360	360;
	13	55	0.02216	0.0904	0.0017	72	72	72	0	0	1	-360	360;
	14	44	0.04211	0.08545	0.0474	81	81	81	0	0	1	-360	360;
	14	34	0.02625	0.06429	0.0012	72	72	72	0	0	1	-360	360;
	15	24	0.01835	0.02845	0.0221	74	74	74	0	0	1	-360	360;
	15	16	0.00753	0.00582	0.0206	91	91	91	0	0	1	-360	360;
	15	54	0.00859	0.00472	0.0214	68	68	68	0	0	1	-360	360;
	15	54	0.00855	0.0047	0.0213	68	68	68	0	0	1	-360	360;
	15	54	0.00846	0.00465	0.021	68	68	68	0	0	1	-360	360;
	16	27	0.00366	0.01312	0.046	112	112	112	0	0	1	-360	360;
	17	19	0.02703	0.03613	0.002	50	50	72	0	0	1	-360	360;
	18	37	0.01017	0.00559	0.0253	68	68	68	0	0	1	-360	360;
	18	37	0.01017	0.00559	0.0253	68	68	68	0	0	1	-360	360;
	20	50	0.0424	0.07509	0.0008	82	82	82	0	0	1	-360	360;
	20	48	0.02133	0.0521	0.0009	82	82	82	0	0	1	-360	360;
	20	34	0.02423	0.05862	0.0011	72	72	72	0	0	1	-360	360;
	21	48	0.01829	0.04246	0.0078	72	72	72	0	0	1	-360	360;
	21	48	0.01824	0.04236	0.0075	72	72	72	0	0	1	-360	360;
	24	44	0.03993	0.09965	0.002	82	82	82	0	0	1	-360	360;
	28	29	0.00087	0.051	0	220	220	220	2.5	0	1	-360	360;
	28	29	0.00087	0.051	0	220	220	220	2.5	0	1	-360	360;
	29	41	0.01868	0.1259	0.0356	191	191	191	0	0	1	-360	360;
	30	41	0.00735	0.04395	0.0123	191	191	191	0	0	1	-360	360;
	30	32	0.00225	0.0134	0.004	191	191	191	0	0	1	-360	360;
	31	28	0.00224	0.03268	0.35	600	600	720	0	0	1	-360	360;
	31	38	0.00075	0.01092	0.117	597	597	597	0	0	1	-360	360;
	32	29	0.0103	0.05681	0.0164	191	191	191	0	0	1	-360	360;
	33	50	0.0502	0.101	0.0025	82	82	82	0	0	1	-360	360;
	33	32	0.0025	0.0723	0	101	101	101	0.5	0	1	-360	360;
	35	56	0.01243	0.07847	0.0192	233	233	233	0	0	1	-360	360;
	35	39	0.01028	0.07253	0.0176	288	288	288	0	0	1	-360	360;
	35	31	0.00087	0.051	0	220	220	220	0.4	0	1	-360	360;
	39	47	0.00775	0.05244	0.0154	233	233	233	0	0	1	-360	360;
	39	40	0.01161	0.08085	0.0225	233	233	233	0	0	1	-360	360;
	39	38	0.00094	0.05107	-0.0127	224	255	224	0.4	0	1	-360	360;
	39	38	0.00095	0.05116	-0.0124	224	255	224	0.4	0	1	-360	360;
	44	41	0.0025	0.07144	-0.0045	101	101	101	0.5	0	1	-360	360;
	44	41	0.00244	0.06829	-0.006	101	101	101	0.5	0	1	-360	360;
	47	53	0.00204	0.00692	0.1601	185	185	185	0	0	1	-360	360;
	48	54	0.01473	0.036	0.0744	83	93	93	0	0	1	-360	360;
	48	47	0.00101	0.03925	-0.0035	187	187	187	0.5	0	1	-360	360;
	54	55	0.06246	0.08246	0.0003	50	50	50	0	0	1	-360	360;
	54	53	0.00134	0.04988	-0.0004	187	187	187	0.5	0	1	-360	360;
	56	55	0.0071	0.0544	0.0138	233	233	233	0	0	1	-360	360;
];

%%-----  OPF Data  -----%%
%% generator cost data
%	1	startup	shutdown	n	x1	y1	...	xn	yn
%	2	startup	shutdown	n	c(n-1)	...	c0
mpc.gencost = [
	2	0	0	3	0.005	36	0;
	2	0	0	3	0.0082	15.2	0;
	2	0	0	3	0.0097	14.5	0;
	2	0	0	3	0.0123	14.3	0;
	2	0	0	3	0.026	25	0;
	2	0	0	3	0.005	28	0;
	2	0	0	3	0.03	22	0;
	2	0	0	3	0.025	16.3	0;
	2	0	0	3	0.023	20	0;
];

fanbigg, 2023-01-20:
I grabbed the data from Donald text book case (example 6.13) but when I ran it in MatPower, it is not converging

closed 2023-01-20 by fanbigg

rdzman, 2023-01-20:
I assume you found the issue?