MATPOWER/matpower #162: power flow not converging

priayad7, 2023-01-31:
I am trying to run power flow analysis in case33bw. I geet convergence error.
Could you please guide me as how should i go about it.

in my case info i see that power dispatch is less than the generarted. and maybe its because of that but how should i change that value ?

function mpc = case33bw
%CASE33BW  Power flow data for 33 bus distribution system from Baran & Wu
%   Please see CASEFORMAT for details on the case file format.
%
%   Data from ...
%       M. E. Baran and F. F. Wu, "Network reconfiguration in distribution
%       systems for loss reduction and load balancing," in IEEE Transactions
%       on Power Delivery, vol. 4, no. 2, pp. 1401-1407, Apr 1989.
%       doi: 10.1109/61.25627
%       URL: https://doi.org/10.1109/61.25627

%% MATPOWER Case Format : Version 2
mpc.version = '2';

%%-----  Power Flow Data  -----%%
%% system MVA base
mpc.baseMVA = 10;

%% bus data
%	bus_i	type	Pd	Qd	Gs	Bs	area	Vm	Va	baseKV	zone	Vmax	Vmin
mpc.bus = [ %% (Pd and Qd are specified in kW & kVAr here, converted to MW & MVAr below)
	1	3	2330    2330*.91	0   0	1	1	0	12.66	1	1	1;
	2	1	2180	2180*.91	0	0	1	1	0	12.66	1	1.1	0.9;
	3	1	2078	2078*.91	0	0	1	1	0	12.66	1	1.1	0.9;
	4	1	2241	2241*.91	0	0	1	1	0	12.66	1	1.1	0.9;
	5	1	2282	2282*.91	0	0	1	1	0	12.66	1	1.1	0.9;
	6	1	2200	2200*.91	0	0	1	1	0	12.66	1	1.1	0.9;
	7	1	2196	2196*.91	0	0	1	1	0	12.66	1	1.1	0.9;
	8	1	2380	2380*.91	0	0	1	1	0	12.66	1	1.1	0.9;
	9	1	2363	2363*.91	0	0	1	1	0	12.66	1	1.1	0.9;
	10	1	2241	2241*.91	0	0	1	1	0	12.66	1	1.1	0.9;
	11	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	12	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	13	1   0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	14	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	15	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	16	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	17	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	18	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	19	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	20	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	21	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	22	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	23	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	24	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	25	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	26	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	27	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	28	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	29	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	30	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	31	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	32	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
	33	1	0	0	0	0	1	1	0	12.66	1	1.1	0.9;
];

%% generator data
%	bus	Pg	Qg	Qmax	Qmin	Vg	mBase	status	Pmax	Pmin	Pc1	Pc2	Qc1min	Qc1max	Qc2min	Qc2max	ramp_agc	ramp_10	ramp_30	ramp_q	apf
mpc.gen = [
	1	10000	0	10	-10	1	100	1	10	0	0	0	0	0	0	0	0	0	0	0	0;
];

%% branch data
%	fbus	tbus	r	x	b	rateA	rateB	rateC	ratio	angle	status	angmin	angmax
mpc.branch = [  %% (r and x specified in ohms here, converted to p.u. below)
	1	2	0.0922	0.0470	0	0	0	0	0	0	1	-360	360;
	2	3	0.4930	0.2511	0	0	0	0	0	0	1	-360	360;
	3	4	0.3660	0.1864	0	0	0	0	0	0	1	-360	360;
	4	5	0.3811	0.1941	0	0	0	0	0	0	1	-360	360;
	5	6	0.8190	0.7070	0	0	0	0	0	0	1	-360	360;
	6	7	0.1872	0.6188	0	0	0	0	0	0	1	-360	360;
	7	8	0.7114	0.2351	0	0	0	0	0	0	1	-360	360;
	8	9	1.0300	0.7400	0	0	0	0	0	0	1	-360	360;
	9	10	1.0440	0.7400	0	0	0	0	0	0	1	-360	360;
	10	11	0.1966	0.0650	0	0	0	0	0	0	1	-360	360;
	11	12	0.3744	0.1238	0	0	0	0	0	0	1	-360	360;
	12	13	1.4680	1.1550	0	0	0	0	0	0	1	-360	360;
	13	14	0.5416	0.7129	0	0	0	0	0	0	1	-360	360;
	14	15	0.5910	0.5260	0	0	0	0	0	0	1	-360	360;
	15	16	0.7463	0.5450	0	0	0	0	0	0	1	-360	360;
	16	17	1.2890	1.7210	0	0	0	0	0	0	1	-360	360;
	17	18	0.7320	0.5740	0	0	0	0	0	0	1	-360	360;
	2	19	0.1640	0.1565	0	0	0	0	0	0	1	-360	360;
	19	20	1.5042	1.3554	0	0	0	0	0	0	1	-360	360;
	20	21	0.4095	0.4784	0	0	0	0	0	0	1	-360	360;
	21	22	0.7089	0.9373	0	0	0	0	0	0	1	-360	360;
	3	23	0.4512	0.3083	0	0	0	0	0	0	1	-360	360;
	23	24	0.8980	0.7091	0	0	0	0	0	0	1	-360	360;
	24	25	0.8960	0.7011	0	0	0	0	0	0	1	-360	360;
	6	26	0.2030	0.1034	0	0	0	0	0	0	1	-360	360;
	26	27	0.2842	0.1447	0	0	0	0	0	0	1	-360	360;
	27	28	1.0590	0.9337	0	0	0	0	0	0	1	-360	360;
	28	29	0.8042	0.7006	0	0	0	0	0	0	1	-360	360;
	29	30	0.5075	0.2585	0	0	0	0	0	0	1	-360	360;
	30	31	0.9744	0.9630	0	0	0	0	0	0	1	-360	360;
	31	32	0.3105	0.3619	0	0	0	0	0	0	1	-360	360;
	32	33	0.3410	0.5302	0	0	0	0	0	0	1	-360	360;
	21	8	2.0000	2.0000	0	0	0	0	0	0	0	-360	360;
	9	15	2.0000	2.0000	0	0	0	0	0	0	0	-360	360;
	12	22	2.0000	2.0000	0	0	0	0	0	0	0	-360	360;
	18	33	0.5000	0.5000	0	0	0	0	0	0	0	-360	360;
	25	29	0.5000	0.5000	0	0	0	0	0	0	0	-360	360;
];

%%-----  OPF Data  -----%%
%% generator cost data
%	1	startup	shutdown	n	x1	y1	...	xn	yn
%	2	startup	shutdown	n	c(n-1)	...	c0
mpc.gencost = [
	2	0	0	3	0	20	0;
];

%% convert branch impedances from Ohms to p.u.
[PQ, PV, REF, NONE, BUS_I, BUS_TYPE, PD, QD, GS, BS, BUS_AREA, VM, ...
    VA, BASE_KV, ZONE, VMAX, VMIN, LAM_P, LAM_Q, MU_VMAX, MU_VMIN] = idx_bus;
[F_BUS, T_BUS, BR_R, BR_X, BR_B, RATE_A, RATE_B, RATE_C, ...
    TAP, SHIFT, BR_STATUS, PF, QF, PT, QT, MU_SF, MU_ST, ...
    ANGMIN, ANGMAX, MU_ANGMIN, MU_ANGMAX] = idx_brch;
Vbase = mpc.bus(1, BASE_KV) * 1e3;      %% in Volts
Sbase = mpc.baseMVA * 1e6;              %% in VA
mpc.branch(:, [BR_R BR_X]) = mpc.branch(:, [BR_R BR_X]) / (Vbase^2 / Sbase);

%% convert loads from kW to MW
mpc.bus(:, [PD, QD]) = mpc.bus(:, [PD, QD]) / 1e3;

priayad7, 2023-01-31:
case info 

Checking connectivity ... single fully connected network
Elapsed time is 0.048787 seconds.
================================================================================
                        Full    
                       System   
Number of:           ---------- 
  buses                    33   
  loads                    10   
    on                     10   
    off                     -   
    fixed                  10   
    dispatchable            -   
      on                    -   
      off                   -   
  generators                1   
    on                      1   
    off                     -   
  shunt elements            -   
  branches                 37   
    on                     32   
    off                     5   
    ties (off)              -   

Load                
  active (MW)       
    dispatched             22.5 
      fixed                22.5 
      dispatchable          -   
    curtailed               -   
    nominal                22.5 
      on                   22.5 
      off                   -   
      fixed                22.5 
      dispatchable          -   
        on                  -   
        off                 -   
  reactive (MVAr)   
    dispatched             20.5 
      fixed                20.5 
      dispatchable          -   
    curtailed               -   
    nominal                20.5 
      on                   20.5 
      off                   -   
      fixed                20.5 
      dispatchable          -   
        on                  -   
        off                 -   

Generation          
  active (MW)       
    dispatched              -   
    max capacity           10.0 
      on                   10.0 
      off                   -   
    min capacity            -   
      on                    -   
      off                   -   
  reactive (MVAr)   
    dispatched              -   
    max capacity           10.0 
      on                   10.0 
      off                   -   
    min capacity          -10.0 
      on                  -10.0 
      off                   -   

Shunt Injections    
    active (MW)             -   
    reactive (MVAr)         -   

Branch Losses       
    active (MW)             -   
    reactive (MVAr)         -   

DC line             
  export (MW)       
    dispatch                -   
    max capacity            -   
      on                    -   
      off                   -   
    min capacity            -   
      on                    -   
      off                   -   

Reference Buses     
  num of ref buses          1   
  ref bus numbers           1

rdzman, 2023-01-31:
First, just for clarity for others who might happen upon this ... this is not the original `case33bw`, but rather a modified version with totally different loads.

I suspect the issue may be related to the fact that you have large parts of the network that are energized, but with no load. 

It looks to me like you've reached a voltage collapse. Try scaling the load down by half (power flow does converge), then use the continuation power flow to explore what happens as you increase the load back to your base level.
```
mpc = loadcase('case_issue162');
mpc_base = scale_load(0.5, mpc);
rb = runpf(mpc_base);
mpopt = mpoption('verbose', 2, 'cpf.plot.level', 2, 'cpf.plot.bus', [8:10]);
r = runcpf(mpc_base, mpc);
```

priayad7, 2023-01-31:
yeah the load values are give by me which I have got from a different test bed, which captures power used by EV. Is that not the correct way to give load as input in order to do power flow analysis ?

rdzman, 2023-01-31:
You have specified the data correctly for MATPOWER to attempt to solve a power flow.

The issue is that **those** loads, in **those** locations on **that** network result in a voltage collapse. So, if those really are the loads you need to serve, you need whoever designed your network to go back and redesign it to handle that particular load pattern.

priayad7, 2023-01-31:
oky so when the load is below 10MW it works fine. Not when it is above 10 MW

rdzman, 2023-01-31:
... if voltages under 0.7 still counts as "works fine" for you.

rdzman, 2023-01-31:
By the way, I forgot to pass the `mpopt` to `runcpf` in the example code above (just edited it now), so it wouldn't create the plot I intended for you to see.

priayad7, 2023-02-01:
mpopt = mpoption('verbose', 2, 'cpf.plot.level', 2, 'cpf.plot.bus', [1:10]); when i run this i get a plot of voltage vs lamba for all the buses, what does it mean by voltage? which voltage is it ?

priayad7, 2023-02-01:
when i scale the load back to its original value it seems to be collapsing again. what should i do to make it run using runcpf?

rdzman, 2023-02-02:
Running the CPF is just a way to illustrate what happens to the voltages in that system when the load is increased from the base system (lambda=0) to the target system (lambda=1). That is, the voltage collapses. In other words, there is no solution to the power flow problem for that system with the target load level. That system cannot physically serve those loads. Under those circumstances, the voltages cannot be maintained without upgrades to the network.

priayad7, 2023-02-08:
Thank you so much for the response. It seems my network cant seem to function at 100% load. My aim is to estimate the line losses. My load data is from small neighborhood, say 100 homes. what network or case is advisable for this type of simulation? i was using case33bw

rdzman, 2023-02-09:
I'm afraid this is beyond the scope of what we can answer here ... getting close to [FAQ #2](https://matpower.org/doc/faq/#homework) territory.

I suggest you find a partner with the necessary power engineering skills to help you design or find a network that is consistent with your intended studies.

closed 2023-02-13 by priayad7