MATPOWER/matpower #163: DCOPF can converge on original matpower case, but cannot converge after applying load2disp

yanda1234, 2023-02-05:
Hi Dr. Zimmerman,

I encountered a question which seems weird to me. The original matpower case can get DCOPF converge but fail after applying load2disp. In my understanding, load2disp will only change load from fixed type to dispatchable type. How could it lead to divergence?
Kindly advise. Many thanks in advance.
[MPC_300.txt](https://github.com/MATPOWER/matpower/files/10611621/MPC_300.txt)

Best,
Yanda

rdzman, 2023-02-06:
Which solver are you using? I suspect it is a numerical issue. Some solvers are less robust than others and I suspect that the solver you are using is encountering numerical issues on the case with dispatchable loads.

I've tried it with several different solvers, and I'm seeing some that fail with numerical difficulty, but other succeed. You are right in assuming that converting loads to dispatchable should not make the problem infeasible, and in fact it does not. But it apparently does make it more numerically difficult for certain algorithms.

yanda1234, 2023-02-06:
I used gurobi based on your suggestion. Which solver works?

rdzman, 2023-02-06:
I'm not sure why, but appears that, if you let Gurobi decide (the default) it selects its barrier method on this problem. Manually selecting its dual simplex (or primal simplex) seems to work just fine.
```
mpopt = mpoption('gurobi.method', 1);
rundcopf(mpc, mpopt);
```
In my experience, the dual simplex is fast and more numerically reliable than barrier. Barrier typically only becomes faster on very large-scale problems. I'm surprised that dual simplex was not the default on this problem.

yanda1234, 2023-02-06:
You are a genius. Thanks, Dr. Zimmerman.

closed 2023-02-06 by yanda1234