MATPOWER/matpower #164: bustypes issue with IEEE-300

irfanbabar42, 2023-02-13:
Hi I am trying to get the information for IEEE-300 bus system with Matpower version 7.1. 
`mpc = loadcase(case300);

[ref, pv, pq] = bustypes(mpc.bus, mpc.gen);
`
Error: 
Error using sparse
Index exceeds array bounds.

Error in bustypes (line 28)
Cg = sparse(gen(:, GEN_BUS), (1:ng)', gen(:, GEN_STATUS) > 0, nb, ng);  %% gen connection matrix

Error in untitled3 (line 7)
[ref, pv, pq] = bustypes(mpc.bus, mpc.gen);

irfanbabar42, 2023-02-14:
@rdzman When can we expect there will be addition to bad data detection for state estimation? Morever, I need help what does it mean at line 41 of the code in https://matpower.org/docs/ref/matpower5.0/extras/se/test_se_14bus.html, "Measurements available". Can we use any measurements at different locations?

rdzman, 2023-02-14:
_(In the future, please open a new issue with a descriptive title when asking a new question.)_

Unfortunately, the state estimation code is contributed code (i.e. not written by me) that was never integrated into the fully supported MATPOWER core distribution. So my ability to support it is very limited. I do not personally have any plans to incorporate bad data detection. However, I always do [welcome contributions](https://github.com/MATPOWER/matpower/blob/master/CONTRIBUTING.md) from the community to improve the code, so feel free to submit any improvements you are able to make. I would even be happy to include state estimation into the fully supported MATPOWER core distribution if someone has the time to clean it up and document it fully.

To your question about line 41 of `test_se_bus.m` ([line 20 of the current version](https://github.com/MATPOWER/matpower-extras/blob/9d276b545a3df0290c881c37a5c830d278527b61/se/test_se_14bus.m#L20)), yes, I believe you can specify measurements at arbitrary locations by specifying the indices (lines 20-28), then the corresponding measurements (lines 30-38) and variances (lines 40-48).

irfanbabar42, 2023-02-15:
@rdzman Thank you. Next time I will try to start a new conversation. Your comments are really helpful.

closed 2023-02-16 by irfanbabar42