MATPOWER/matpower #166: Error while installing Matpower

DiegoCanas, 2023-03-01:
I'm trying to install Matpower, on installation Options y answered number 3 "   3. DO modify the MATLAB path, and SAVE the updated path (so you will not have to do it again next time you run MATLAB)", then I run the Matpower tests and I always get the same error:

t_psse....................Error using cellfun
Non-scalar in Uniform output, at index 2, output 1.
Set 'UniformOutput' to false.
Error in t_run_tests (line 61)
    feval( test_names{k}, ~verbose );
Error in test_matpower (line 194)
all_ok = t_run_tests( tests, verbose );
Error in install_matpower (line 394)
            test_matpower;

[image]

I´m aiming to convert a PSSE .raw file to Matpower
What should I do? Thank you.

rdzman, 2023-03-01:
I need a few more details to be able to begin tracking down the problem.

- Which version of MATPOWER?
- And which version of MATLAB?

Maybe just post the output of the `mpver` command.

rdzman, 2023-03-01:
Could you also post the full output of the `t_psse` command?

DiegoCanas, 2023-03-02:
Hi! The versions: 

[image]

[image]

The results of t_psse:

>> t_psse
1..143
ok 1 - [records, sections] = psse_read() : length(records)
ok 2 - [records, sections] = psse_read() : length(sections)
ok 3 - psse_parse_line(str1, template) : length
ok 4 - psse_parse_line(str1, template) :  col 1
ok 5 - psse_parse_line(str1, template) :  col 2
ok 6 - psse_parse_line(str1, template) :  col 3
ok 7 - psse_parse_line(str1, template) :  col 4
ok 8 - psse_parse_line(str1, template) :  col 5
ok 9 - psse_parse_line(str1, template) :  col 6
ok 10 - psse_parse_line(str1, template) :  col 7
ok 11 - psse_parse_line(str1, template) :  col 8
ok 12 - psse_parse_line(str1, template) :  col 9
ok 13 - psse_parse_line(str1, template) :  col 10
ok 14 - psse_parse_line(str1, template) :  comment
ok 15 - psse_parse_line(str2, template) : length
ok 16 - psse_parse_line(str2, template) :  col 1
ok 17 - psse_parse_line(str2, template) :  col 2
ok 18 - psse_parse_line(str2, template) :  col 3
ok 19 - psse_parse_line(str2, template) :  col 4
ok 20 - psse_parse_line(str2, template) :  col 5
ok 21 - psse_parse_line(str2, template) :  col 6
ok 22 - psse_parse_line(str2, template) :  col 7
ok 23 - psse_parse_line(str2, template) :  col 8
ok 24 - psse_parse_line(str2, template) :  col 9
ok 25 - psse_parse_line(str2, template) :  col 10
ok 26 - psse_parse_line(str2, template) :  comment
ok 27 - psse_parse_line(str3, template) : length
ok 28 - psse_parse_line(str3, template) :  col 1
ok 29 - psse_parse_line(str3, template) :  col 2
ok 30 - psse_parse_line(str3, template) :  col 3
ok 31 - psse_parse_line(str3, template) :  col 4
ok 32 - psse_parse_line(str3, template) :  col 5
ok 33 - psse_parse_line(str3, template) :  col 6
ok 34 - psse_parse_line(str3, template) :  col 7
ok 35 - psse_parse_line(str3, template) :  col 8
ok 36 - psse_parse_line(str3, template) :  col 9
ok 37 - psse_parse_line(str3, template) :  col 10
ok 38 - psse_parse_line(str3, template) :  comment
ok 39 - psse_parse_line(str4, template) : length
ok 40 - psse_parse_line(str4, template) :  col 1
ok 41 - psse_parse_line(str4, template) :  col 2
ok 42 - psse_parse_line(str4, template) :  col 3
ok 43 - psse_parse_line(str4, template) :  col 4
ok 44 - psse_parse_line(str4, template) :  col 5
ok 45 - psse_parse_line(str4, template) :  col 6
ok 46 - psse_parse_line(str4, template) :  col 7
ok 47 - psse_parse_line(str4, template) :  col 8
ok 48 - psse_parse_line(str4, template) :  col 9
ok 49 - psse_parse_line(str4, template) :  col 10
ok 50 - psse_parse_line(str4, template) :  comment
ok 51 - psse_parse_line : missing optional columns : length
ok 52 - psse_parse_line : missing optional columns : all empty
ok 53 - psse_parse_section : struct
ok 54 - psse_parse_section : size(num)
ok 55 - psse_parse_section : size(txt)
ok 56 - psse_parse_section :  num(1,1)
ok 57 - psse_parse_section :  txt{1,1}
ok 58 - psse_parse_section :  num(1,2)
ok 59 - psse_parse_section :  txt{1,2}
ok 60 - psse_parse_section :  num(1,3)
ok 61 - psse_parse_section :  txt{1,3}
ok 62 - psse_parse_section :  num(1,4)
ok 63 - psse_parse_section :  txt{1,4}
ok 64 - psse_parse_section :  num(1,5)
ok 65 - psse_parse_section :  txt{1,5}
ok 66 - psse_parse_section :  num(1,6)
ok 67 - psse_parse_section :  txt{1,6}
ok 68 - psse_parse_section :  num(1,7)
ok 69 - psse_parse_section :  txt{1,7}
ok 70 - psse_parse_section :  num(1,8)
ok 71 - psse_parse_section :  txt{1,8}
ok 72 - psse_parse_section :  num(1,9)
ok 73 - psse_parse_section :  txt{1,9}
ok 74 - psse_parse_section :  num(1,10)
ok 75 - psse_parse_section :  txt{1,10}
ok 76 - psse_parse_section :  num(2,1)
ok 77 - psse_parse_section :  txt{2,1}
ok 78 - psse_parse_section :  num(2,2)
ok 79 - psse_parse_section :  txt{2,2}
ok 80 - psse_parse_section :  num(2,3)
ok 81 - psse_parse_section :  txt{2,3}
ok 82 - psse_parse_section :  num(2,4)
ok 83 - psse_parse_section :  txt{2,4}
ok 84 - psse_parse_section :  num(2,5)
ok 85 - psse_parse_section :  txt{2,5}
ok 86 - psse_parse_section :  num(2,6)
ok 87 - psse_parse_section :  txt{2,6}
ok 88 - psse_parse_section :  num(2,7)
ok 89 - psse_parse_section :  txt{2,7}
ok 90 - psse_parse_section :  num(2,8)
ok 91 - psse_parse_section :  txt{2,8}
ok 92 - psse_parse_section :  num(2,9)
ok 93 - psse_parse_section :  txt{2,9}
ok 94 - psse_parse_section :  num(2,10)
ok 95 - psse_parse_section :  txt{2,10}
ok 96 - psse_parse_section :  num(3,1)
ok 97 - psse_parse_section :  txt{3,1}
ok 98 - psse_parse_section :  num(3,2)
ok 99 - psse_parse_section :  txt{3,2}
ok 100 - psse_parse_section :  num(3,3)
ok 101 - psse_parse_section :  txt{3,3}
ok 102 - psse_parse_section :  num(3,4)
ok 103 - psse_parse_section :  txt{3,4}
ok 104 - psse_parse_section :  num(3,5)
ok 105 - psse_parse_section :  txt{3,5}
ok 106 - psse_parse_section :  num(3,6)
ok 107 - psse_parse_section :  txt{3,6}
ok 108 - psse_parse_section :  num(3,7)
ok 109 - psse_parse_section :  txt{3,7}
ok 110 - psse_parse_section :  num(3,8)
ok 111 - psse_parse_section :  txt{3,8}
ok 112 - psse_parse_section :  num(3,9)
ok 113 - psse_parse_section :  txt{3,9}
ok 114 - psse_parse_section :  num(3,10)
ok 115 - psse_parse_section :  txt{3,10}
ok 116 - psse_parse_section :  num(4,1)
ok 117 - psse_parse_section :  txt{4,1}
ok 118 - psse_parse_section :  num(4,2)
ok 119 - psse_parse_section :  txt{4,2}
ok 120 - psse_parse_section :  num(4,3)
ok 121 - psse_parse_section :  txt{4,3}
ok 122 - psse_parse_section :  num(4,4)
ok 123 - psse_parse_section :  txt{4,4}
ok 124 - psse_parse_section :  num(4,5)
ok 125 - psse_parse_section :  txt{4,5}
ok 126 - psse_parse_section :  num(4,6)
ok 127 - psse_parse_section :  txt{4,6}
ok 128 - psse_parse_section :  num(4,7)
ok 129 - psse_parse_section :  txt{4,7}
ok 130 - psse_parse_section :  num(4,8)
ok 131 - psse_parse_section :  txt{4,8}
ok 132 - psse_parse_section :  num(4,9)
ok 133 - psse_parse_section :  txt{4,9}
ok 134 - psse_parse_section :  num(4,10)
ok 135 - psse_parse_section :  txt{4,10}

[image]

rdzman, 2023-03-02:
It looks like some of your MATPOWER files have been modified. In particular, line 109 of `t_psse.m` in your error message does not match [the original](https://github.com/MATPOWER/matpower/blob/ad9738b9c0bc794c84dea5db399e9a6496ca8bd5/lib/t/t_psse.m#L109).

I suggest you try deleting and reinstalling MATPOWER to ensure you are using an unmodified copy. Then re-run the tests and report back.

rdzman, 2023-03-03:
That installer command does not delete the modified files. It just removes MATPOWER from your Matlab path.

The issue is that what is installed in your MATPOWER folder is no longer an unmodified MATPOWER, it's been modified in ways that are unknown to me. To fix that, you will need to actually delete the MATPOWER directory and reinstall all of the files from the original source in order to ensure you have a clean, unmodified MATPOWER.

**Note:** You should **never** have to modify any files that came with MATPOWER in order to use it.

DiegoCanas, 2023-03-06:
Hi! Thank you! Now I don't get any error while installling matpower but I get another error when I try to use the psse2mpc, should I open a new issue?
This is the code that I write:
[image]

And this is the error that i get:
[image]
Thank you!

rdzman, 2023-03-07:
👍

closed 2023-04-11 by rdzman