MATPOWER/matpower #168: How to add some new constraints in Matpower

lzq-zbc, 2023-03-06:


rdzman, 2023-03-07:
See Section 6.3 and Chapter 7 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf).

Feel free to ask a more specific question if there is a particular place you are getting stuck.

lzq-zbc, 2023-03-14:
Having some problems reading om.add_lin_constraint('myLinCons', A, l, u, varsets);  ，varsets represents some variables.However,in the ACOPF model that runs case9.m,how should I use varsets to represent the three variables of the active output of the generator set in the ACOPF objective function.

rdzman, 2023-03-20:
It looks like you are mixing the two approaches to customizing the OPF. If you are using direct specification, you simply add the `A`, `l`, `u`, `Q`, `c`, `k` as fields in `mpc`, then call `runopf()` with that `mpc`. But as I said, for direct specification, you will need to provide parameters that are consistent with the full optimization vector `x`, which consists of voltage angles and magnitudes first, then active and reactive generation. So, for the 9-bus case, the first 18 elements of `x` (and correspondingly first 18 columns of `A`) are voltages, followed by 3 elements for active generation and 3 for reactive.

To call methods on the `om` to add variables, costs, constraints, you will (generally) have to do that from within a callback function as illustrated in Chapter 7 of the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf).

Maybe I can give you a hand if you tell me exactly what is the constraint you are trying to add?

lzq-zbc, 2023-03-21:
Thanks for your reply, I did have some problems.
My intention was to add a new set of variables, constraints, and costs to the case9 study model. The number of variables in this group is equal to the number of generators, this variable is named Cg, and the constraint of this set of variables is a linear inequality, l<=A1*Cg1+A2*Cg2+A3*Cg3<=u,A1,A2,A3 represents the coefficients of the variables. Finally, this set of variables is added to the cost, and the added cost is a quadratic function w1i*Cgi²+w2i*Cgi+w3i, i represents the ordinal number of the variable, w1i, w2i, w3i represent the quadratic term coefficient, primary term coefficient, constant in the cost of the ith variable, respectively.
Also, the numbers in the above program are all randomly defined by me, and I would appreciate it if you could reply to me.

lzq-zbc, 2023-03-23:
First of all, thank you very much, Mr. Zimmerman. 
Since I'm not very proficient in matpower, I'd like to just add a few simple constraints (which really have nothing to do with the original OPF model) to increase my proficiency with matpower, and also to figure out how to use the add constraint and add cost program statements, but I don't know if that's possible, and I'd appreciate it if you'd like to help me

rdzman, 2023-03-23:
There is an example in the OPF tests (e.g. starting at [line 228](https://github.com/MATPOWER/matpower/blob/a366e18b8bd49ebb606ac078505c96a570a138c2/lib/t/t_opf_default.m#L228) of `t_opf_default`) that defines 1 extra variable (**z**), 2 constraints, and a cost on the new variable. The new variable is constrained to be greater than or equal to the difference between the bus 1 voltage magnitude and 1 p.u. And the cost on that deviation from 1 p.u. is $100.

There are 24 variables to begin with, 9 voltage angles, 9 voltage magnitudes, 3 active gen injections, and 3 reactive gen injections. So defining **A** with 25 columns introduces the **z** variable as column 25, with column 10 corresponding to the voltage magnitude at bus 1.

Here is the code to run that example standalone.
```
mpc = loadcase('t_case9_opf');
mpc.A = sparse([1;1;2;2],[10;25;10;25],[-1;1;1;1],2,25);
mpc.u = [Inf; Inf];
mpc.l = [-1; 1];
mpc.N = sparse(1, 25, 1, 1, 25);    % w = new z variable only
mpc.Cw = 100;
r = runopf(mpc);
z = r.var.val.z
new_cost = full(r.cost.usr)
```

**Note:** _This is using the direct specification method of modifying the OPF formulation. Using callbacks and the add constraint and add cost methods is more involved._

lzq-zbc, 2023-04-05:
Through this period of learning, I have gained a certain understanding of the formula of using direct specification methods to modify OPF. But I'm also interested in another approach, using callbacks and adding constraints and adding costs, and I wonder if you have a complete and simple example for me to draw on, for which I would appreciate it

lzq-zbc, 2023-04-13:
This is a piece of code I generated with chatgpt to add a nonlinear constraint to the case30 study.

### 
function mpc = add_nonlin_constraint(mpc)
%ADD_NONLIN_CONSTRAINT  Adds a non-linear constraint to the OPF problem.
%   MPC = ADD_NONLIN_CONSTRAINT(MPC) adds a non-linear constraint to the OPF
%   problem defined in MPC. The constraint is defined as x1*x2 <= 0.96, where
%   x1 and x2 are the values of the first and second columns of the matrix
%   MPC.bus, respectively.
%
%   The new constraint is added using the 'formulation' callback. This
%   callback adds the constraint to the problem formulation and updates the
%   appropriate fields of the OPF model struct. The 'formulation' callback is
%   called by MATPOWER's 'opf_setup' function.
%
%   Note: This function only adds a single non-linear constraint to the
%   problem. To add multiple non-linear constraints, this function can be
%   modified or called multiple times.
%
%   See also OPF_SETUP, USERFCN_NONLIN_CONSTRAINT_FORMULATION.

%% define non-linear constraint
nonlin_constraint = struct();
nonlin_constraint.idx = [31; 32];
nonlin_constraint.fcn = @(x) x(31)*x(32) - 0.96;
nonlin_constraint.grad = @(x) [x(32); x(31)];

%% add non-linear constraint formulation callback
mpc = add_userfcn(mpc, 'formulation', @userfcn_nonlin_constraint_formulation);

%% define userfcn_nonlin_constraint_formulation callback function
function om = userfcn_nonlin_constraint_formulation(om)
    x0 = om.var.val;
    fcn = nonlin_constraint.fcn;
    grad = nonlin_constraint.grad;

    %% add non-linear constraint to problem formulation
    om = add_nln_constraint(om, 'nonlin_constraint', nonlin_constraint.idx, 0, fcn, grad, x0);

    %% update variable and constraint indices in MPC
    mpc.new_var_idx = om.var.idx.i1;
    mpc.new_lin_idx = om.con.idx.i1;
end

end
### 

mpc = loadcase('case30');
mpc = add_nonlin_constraint(mpc);
results = runopf(mpc);
###
In the end, an error is reported
####
Incorrect use of add_nonlin_constraint/userfcn_nonlin_constraint_formulation
Too many input parameters.

Error run_userfcn (line 32)
        rv = feval(userfcn.( stage)(k).fcn, rv, varargin{2:end}, args);

Error opf_setup (line 514)
om = run_userfcn(userfcn, 'formulation', om, mpopt);

Error opf (line 225)
om = opf_setup(mpc, mpopt);

Error runopf (line 75)
[r, success] = opf(casedata, mpopt);

Error test3 (line 3)
results = runopf(mpc);
#####
In my opinion, this model structure should be complete, but I don't know where the problem arises

lzq-zbc, 2023-04-13:
This is a piece of code I generated with chatgpt to add a nonlinear constraint to the case30 study.
`function mpc = add_nonlin_constraint(mpc)
%ADD_NONLIN_CONSTRAINT Adds a non-linear constraint to the OPF problem.
% MPC = ADD_NONLIN_CONSTRAINT(MPC) adds a non-linear constraint to the OPF
% problem defined in MPC. The constraint is defined as x1*x2 <= 0.96, where
% x1 and x2 are the values of the first and second columns of the matrix
% MPC.bus, respectively.
%
% The new constraint is added using the 'formulation' callback. This
% callback adds the constraint to the problem formulation and updates the
% appropriate fields of the OPF model struct. The 'formulation' callback is
% called by MATPOWER's 'opf_setup' function.
%
% Note: This function only adds a single non-linear constraint to the
% problem. To add multiple non-linear constraints, this function can be
% modified or called multiple times.
%
% See also OPF_SETUP, USERFCN_NONLIN_CONSTRAINT_FORMULATION.

%% define non-linear constraint
nonlin_constraint = struct();
nonlin_constraint.idx = [31; 32];
nonlin_constraint.fcn = @(x) x(31)*x(32) - 0.96;
nonlin_constraint.grad = @(x) [x(32); x(31)];

%% add non-linear constraint formulation callback
mpc = add_userfcn(mpc, 'formulation', @userfcn_nonlin_constraint_formulation);

%% define userfcn_nonlin_constraint_formulation callback function
function om = userfcn_nonlin_constraint_formulation(om)
x0 = om.var.val;
fcn = nonlin_constraint.fcn;
grad = nonlin_constraint.grad;

%% add non-linear constraint to problem formulation
om = add_nln_constraint(om, 'nonlin_constraint', nonlin_constraint.idx, 0, fcn, grad, x0);

%% update variable and constraint indices in MPC
mpc.new_var_idx = om.var.idx.i1;
mpc.new_lin_idx = om.con.idx.i1;
end

end`

Here are the instructions I used in the command line window

`mpc = loadcase('case30');
mpc = add_nonlin_constraint(mpc);
results = runopf(mpc);`

In the end, an error is reported

Incorrect use of add_nonlin_constraint/userfcn_nonlin_constraint_formulation
Too many input parameters.

Error run_userfcn (line 32)
rv = feval(userfcn.( stage)(k).fcn, rv, varargin{2:end}, args);

Error opf_setup (line 514)
om = run_userfcn(userfcn, 'formulation', om, mpopt);

Error opf (line 225)
om = opf_setup(mpc, mpopt);

Error runopf (line 75)
[r, success] = opf(casedata, mpopt);

Error test3 (line 3)
results = runopf(mpc);

In my opinion, this model structure should be complete, but I don't know where the problem arises

Finally, I would like to send my heartfelt thanks,Mr. Rdzman

rdzman, 2023-04-13:
First, I don't recommend you trust ChatGPT to "understand" how to use MATPOWER. I don't have time to look through the whole code example, but the use of the `add_nln_constraint()` method is clearly incorrect.  

Please see Section 5.2.2 in the [MP-Opt-Model User's Manual](https://matpower.org/docs/MP-Opt-Model-manual-4.1.pdf) for information on the correct way to call the `add_nln_constraint()` method. In particular, in your case `N` will be 1, since it's a single constraint and you will need to provide function handles for two functions, the first that evaluates the constraint and it's Jacobian, and the second that evaluates the Hessian. Whether or not you use `varsets` is up to you. It depends whether you want your constraint and Hessian functions to receive as an input the entire `x` vector or only a subset of `x`.

lzq-zbc, 2023-05-16:
Mr. Zimmerman，I would like to ask you a question about your published paper`MATPOWER: Steady-State Operations,Planning, and Analysis Tools for Power Systems Research and Education`in 2011.In the example `case118` mentioned in the paper, whether the comparison of that version of Matpower and the current Matpower 8.0 has been changed.

rdzman, 2023-05-16:
The version of `case118` referenced in the 2011 paper I believe is from MATPOWER 4.0b3. It is essentially equivalent to the version in MATPOWER 8.0b1, with the following minor differences:
- branch flow constraints (`RATE_A`) were changed from 9900 MVA to 0 (unconstrained)
- two branches had the `TAP` column changed from 0, indicating a transmission line, to 1, indicating a transformer with nominal tap ratio, which produces no change whatsoever to the resulting admittance matrices
- generator costs include a few more significant digits
- bus names have been added

So, the solutions to the 2 cases should be identical, although the number of constraints included in the OPF is smaller since the (non-binding) line constraints at 9900 MVA were eliminated completely.

closed 2024-04-16 by rdzman