MATPOWER/matpower #169: Using Matpower to find capability curves of distribution grid using AC OPF

akhtar3100, 2023-03-06:
I am trying to find the reactive power capability curves of the distribution grid. To solve this problem, I am using an OPF based formulation where my objective is to find maximum and minimum value at low voltage side of the MV/LV transformer. The objective function is
Obj min,max Qpcc
s.t 
active power balance at each bus
reactive power balance at each bus
and voltage limits  Umin<U<Umax

My question is how can I write this objective function in matpower. The built in objective function in matpower is to minimize generation costs, how can I delete the existing objective function and replace it with the new one. If you have an example code or similar implementation code, can you share it with me?

rdzman, 2023-03-07:
If the objective is to minimize/maximize a simple reactive injection (something you can model with a generator), then it's a simple matter of putting a cost (positive for minimization, negative for maximization) on that reactive generation and setting all other gen costs to zero.

If you need to model the transformer and it is a reactive **flow** you are trying to minimize/maximize, that is more complicated, since the flow is not an explicit variable, but rather a non-linear function of the complex voltages. In that case, I would suggest defining a new variable for the flow, defining a non-linear constraint that equals the flow variable and the flow as computed from the voltages (defined as a nonlinear user constraint, see Sections 6.3 and Chapter 7).

akhtar3100, 2023-03-08:
Hi Zimmerman,

My objective is not to maximize or minimize reactive power of generators. My objective is to find the maximum and minimum values of reactive power of a 'distribution network' at the MV/LV interface with changing values of active power(active power of loads/generatiors) and voltage . It is similar to finding the hosting capacity of network. For that purpose, how should I change my objective function in matpower?

abhyshr, 2023-03-08:
Looks like what you are trying to do here can be done using the continuation power flow functionality `runcpf` in MATPOWER. Using the continuation power flow, you can progressively load a system to its steady-state loading limit. You can ask `runcpf` to stop at the steady-state loading limit (nose point) or when some voltage or flow limit gets hit. `runcpf` stores the output at each load increment that you can use to draw VQ curves, etc. Refer to chapter 5 in the [manual](https://matpower.org/docs/MATPOWER-manual.pdf).

akhtar3100, 2023-03-09:
> Looks like what you are trying to do here can be done using the continuation power flow functionality `runcpf` in MATPOWER. Using the continuation power flow, you can progressively load a system to [...]

I have tried that but it draws V,lambda curves by default. How can I change the code to get the P,Q curves?

akhtar3100, 2023-03-09:
> That is good idea to replace the transformer with the dummy generator and then get the maximum and minimum values of reactive power of that generator. Regarding loads and cost function, I am not modelling them as dispatchable. I am new to matpower, I have no idea how load can be modelled as dispatchable loads in matpower and how should I modify costs to dominate for dummy generator. Can you share an example code? Are there any beginner's course for matpower for quick learning?

rdzman, 2023-03-09:
I'm afraid we don't (yet) have the kind of MATPOWER beginner courses that would be ideal.

Dispatchable loads are discussed in Section 6.4.2 of the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf), and there is also the [`load2disp()`](https://github.com/MATPOWER/matpower/blob/master/lib/load2disp.m) function that can automate the conversion of fixed loads to dispatchable loads.

In terms of costs, I was thinking of something like setting the value of the load to $10, the cost of active power generation (all gens) to $1, and the cost of the reactive generation at the dummy generator to $100 for minimization and -$100 for maximization of reactive injection, with reactive costs of all other generation set to zero.

This should result in dispatching all loads and generators within their PMIN-PMAX and QMIN-QMAX ranges and voltages within their VMIN-VMAX ranges to maximize or minimize the value of reactive injection at the dummy generator.

In case you were not aware, for a case with _ng_ generators, the first _ng_ rows of `mpc.gencost` define the cost of active power, and if you add another _ng_ rows (same format), they define reactive power costs.

The only potential complication I see with this approach is related to the definition and modeling of dispatchable loads. A dispatchable load is modeled as a (negative) generator with a constant power factor constraint added, to force the reactive load to scale with the active load. The issue is that the criteria used to determine if a generator should be treated as a dispatchable load (i.e. the constant power factor constraint should be added), is that it must have PMIN<0 and PMAX==0. In your case, I assume you may **not** want to allow the load to be curtailed all the way to zero (corresponding to PMAX==0).

If that does turn out to be an issue, you should be able to solve it by using piecewise linear costs for the dispatchable loads, with a value of say $1000 on the segment you do not want to be curtailed, and $10 on the portion corresponding to the desired dispatch range.

closed 2023-04-11 by rdzman