MATPOWER/matpower #171: Linear Shift Factors larger than 1

rrnaumann, 2023-03-14:
Hi Ray, 

I've noticed that the PTDF matrix generated by `makePTDF` sometimes displays entries with absolute value larger than 1.

Two cases where this happens: **case145.m** and **case300.m**.

Apparently, this occurs when Branch Data (mpc.branch) contains **negative values** for the branch's reactance (`BR_X`) in column 4. 

When I modify the case data as ` mpc.branch(:,4) = abs(mpc.branch(:,4))`, I get PTDF with absolute value no greater than 1. 

The current code for `makeBdc` is sensitive to the sign of the reactance: `b = stat ./ branch(:, BR_X); %% series susceptance`

Intuitively, I would expect PTDFs to be between -1 and 1. However, I'm not an electrical engineer so I'm opening this issue to make sure that only the magnitude (and not the sign) of the branch reactance matters for real power flows. 

Thanks!

rdzman, 2023-03-14:
The sign of the reactance also matters.

I believe that for real-world, physical lines the reactance is always positive. If I'm not mistaken, negative reactances in power flow models typically stem from using network equivalencing techniques reduce the size of a model.

rrnaumann, 2023-03-16:
Thanks, Ray. I think this answers my question about the code (The sign of `BR_X` matters).

However, before closing the issue, do you have an idea on how to interpret linear shift factors with magnitude larger than 1?

They do show up in real cases. For instance, `case1951rte.m`, `case3375wp.m`, and `case6468rte.m`. 

I'm working with a real network model where a DC OPF is run by the power system operator to allocate transmission rights. It's done using linear shift factors (and not voltage angles). However, the shift factors I get when I use the function `makePTDF` are sometimes outside the unit circle. I then notice that this also happens to a few cases available in MATPOWER. 

I'm having a hard time interpreting these shift factors physically. (How can the change in a branch flow exceed whatever was injected at a bus and extracted at the slack? Is it because existing branch flows can also change completely in response to a new injection and these changes add up? In that sense, is the PTDF matrix always conditional on the existing flows? I originally thought that only the network topology and branch impedances mattered for the PTDF).   

At the same time, it seems that there is nothing in the mathematical definition of the PTDF matrix (Equation 4.36 of MATPOWER's manual) indicating that $-1 \leq PTDF_{i,j} \leq 1$ holds for all real `BR_X`. It could be that these networks are not physically consistent. 

I can move these questions to the discussion list given that there is no code issue here.

rdzman, 2023-03-17:
I'm not 100% certain of this, but I believe that PTDFs with absolute values greater than 1 also arise from equivalencing. I think there is often some equivalencing that is done even in these real-world models included in MATPOWER.

closed 2023-03-17 by rrnaumann