MATPOWER/matpower #173: DCOPF result

yanda1234, 2023-03-20:
Hi Dr. Zimmerman,

While running DCOPF of my own case, I noticed a weird thing. **Changing reference bus from one bus to another** or **changing the order of MPC.bus** will change the branch flows in DCOPF result. I understand my case is kind of large (around 67447 bus and 80000 branches), but is it reasonable these things can happen?
Kindly advise. Many thanks in advance.

Best,
Yanda

rdzman, 2023-03-20:
I don't believe MATPOWER ever changes the reference bus for a DC OPF.

However, internally, it does renumber the buses with consecutive "internal" bus numbers before doing the calculation, then convert back to the original bus numbers to report the results.

Are you saying that the `bus` matrix in the result returned by `runopf()` shows a different reference bus than in the `mpc` you passed in?

yanda1234, 2023-03-20:
Hi Dr. Zimmerman,

Thanks for your prompt reply. Nope, what I meant was if we change reference bus of a MPC case or changing the order of buses in MPC.bus, then we might get different branch flow results after running DCOPF. 

Best,
Yanda

rdzman, 2023-03-20:
I'm sorry. I see I didn't read your question carefully.

So, to answer the question you actually asked ... yes, I am not surprised to see different branch flows (and presumably different generator dispatches) as long as the objective function value is essentially identical. There may be multiple optimal solutions to the problem and **any** change in inputs could result in a problem that is theoretically 100% equivalent, but is not identical numerically, so it ends up at a different local optimum.

Are you seeing the same objective function value for the different cases?

I expect that adding a very small amount of noise to the generator costs could make the solution unique, in which case changing the reference bus or bus ordering should not affect the solution.

yanda1234, 2023-03-20:
> I expect that adding a very small amount of noise to the generator costs could make the solution unique, in which case changing the reference bus or bus ordering should not affect the solution.

Hi Dr. Zimmerman,

Thanks for your detailed reply.
Do you mean adding a very small amount of noise can make changing the reference bus or bus ordering won't affect the solution anymore? If so, could you please tell me how to do that?

Best,
Yanda

abhyshr, 2023-03-20:
Could this behavior be a result of the fixed reference bus angle constraint : $\theta_i = \theta_i^{ref}, i \in \mathcal{I}_{ref}$ ? Changing the reference bus would mean angle at some other bus would be fixed resulting in different bus angles and dispatch(?)

yanda1234, 2023-03-20:
> Could this behavior be a result of the fixed reference bus angle constraint : θi=θiref,i∈Iref ? Changing the reference bus would mean angle at some other bus would be fixed resulting in different bus angles and dispatch(?)

Hi abhyshr,

Do you mean there are some branches connected with reference bus via phase shifters and thus make branch flow change at the end? I think this possibility exists, but in my case, the issue still exists after I removed all phase shifters. 
The new local optimum and previous local optimum did have the same optimal value, so I think Dr. Zimmerman's intuition is right. Probably I need to add some other constraints to avoid moving to the new local optimal point.

rdzman, 2023-03-20:
> Do you mean adding a very small amount of noise can make changing the reference bus or bus ordering won't affect the solution anymore? If so, could you please tell me how to do that?

Exactly. For example, if you have all linear generator costs with the linear coefficient in column 5 of `gencost`, you could modify your case slightly by doing something like ...
```
ng = size(mpc.gen, 1);
cost_noise  = 1e-6 * rand(ng, 1);
mpc.gencost(:, 5) = mpc.gencost(:, 5) + cost_noise;
```

This updated case is much more likely to have a unique solution that will not change with bus ordering or reference changes.

rdzman, 2023-03-20:
> Could this behavior be a result of the fixed reference bus angle constraint?

Changing to a new reference bus could certainly change the voltage angle solutions, but all of the voltage angle **differences** between buses, and therefore flows, should be identical ... even with phase shifters, since they also only care about angle **differences**.