MATPOWER/matpower #175: how to identify relaxed constraints?

yanda1234, 2023-03-30:
Hi Dr. Zimmerman,

I encountered a power flow case that could not be converged using DCOPF. However, after using toggle_softlims to relax some constraints, the DCOPF problem was able to converge successfully. The problem is when I tried to use commands below to identify which constraints are relaxed, I found none of them is relaxed. Do you know how could this happen? If none constraints needs to be relaxed, then why cannot original case get converged?
DCopf.softlims.PMAX.overload~=0,
DCopf.softlims.PMIN.overload~=0,
DCopf.softlims.RATE_A.overload~=0

Kindly advise.

Best,
Yanda

rdzman, 2023-03-31:
I agree that this should not happen. If there are no overloads, then the lack of convergence is not due to infeasibility, but probably some numerical issue with that particular problem and whatever solver is being used.

Which solver are you using?

If the problem size is not huge and the costs are linear, I would recommend selecting a simplex solver, if you have one available. If you are an academic, you should be able to get a [Gurobi](https://www.gurobi.com/academia/academic-program-and-licenses/) license for free, which is typically my solver of choice for DC OPF problems. [MOSEK](https://www.mosek.com/products/academic-licenses/) is another good option. And `linprog` from the MATLAB Optimization Toolbox is also not bad. All are much better than the included MIPS solver for DC OPF problems.

rdzman, 2023-03-31:
Is the case one you can share? If so, feel free to send it to me directly if you don't want to post it here.

rdzman, 2023-03-31:
Could you try the following with your result ...
```
sum([DCopf.softlims.PMAX.ovl_cost; DCopf.softlims.PMIN.ovl_cost; DCopf.softlims.RATE_Aovl_cost])
```
... just to double-check that it also returns 0.

If it does, then maybe you could try relaxing the constraints manually, but just slightly, to see if it will solve without the soft constraints. I'm thinking of something like ...
```
epsilon = 1e-8;
mpc.branch(:, RATE_A) = mpc.branch(:, RATE_A) + epsilon;
mpc.gen(:, PMAX) = mpc.gen(:, PMAX) + epsilon;
mpc.gen(:, PMIN) = mpc.gen(:, PMIN) - epsilon;
```
If that converges, it would confirm that it is a numerical issue. Maybe the feasible region is degenerate in a particular dimension and numerical rounding makes it appear infeasible?

closed 2024-04-16 by rdzman