MATPOWER/matpower #178: Overloaded branches alert/print in ACPF?

raheem-cs, 2023-04-25:
Hello,

I'm trying to see overloaded branches based on rateA when running ACPF (NR), however; such information is not given in the result of the power flow study. I'm searching for options also, but can't see any option to display this information. I'm aware that in ACOPF, such data is printed in results, but I need to know how to show it also in regular AC power flow studies.

Any idea, or what I'm missing?

rdzman, 2023-04-25:
Check out [`checklimits()`](https://github.com/MATPOWER/matpower-extras/blob/master/misc/checklimits.m) in the `extras/misc` subdirectory. This really should be clean up, documented and included in the main distribution, but it's never quite made it to the top of the to-do list.

raheem-cs, 2023-04-25:
Thank you for responding.

I was searching in the main library ([here](https://matpower.org/docs/ref/matpower7.1/lib/menu7.1.html)), hadn't thought that it is not existing or it might exist in extras.

I tried the function on both regular AC power flow and optimal AC power flow, it is not needed in the ACOPF algorithm, because Voltage Constraints, Generation Constraints, and Branch Flow Constraints were already printed out after solving the problem. Hence, it may be added as an extra output option for AC/DC regular power flow to give accessibility for users to see violations as part of the ACPF/DCPF solution.