MATPOWER/matpower #180: Encountering Issues with generation Scheduling in MOST Software: Need Help with High-Probability Contingencies

Mars0077, 2023-04-30:
I'm using the MOST software to schedule power generation for a load profile and applying contingencies with high probabilities of 0.8. I followed the user manual example (ex_case3b) and disabled the wind power codes. While the codes work well for small contingencies, I'm encountering issues with the results when I increase the contingency probabilities beyond 0.3. Specifically, I'm seeing negative generation and negative load dispatching, which doesn't make sense. Can you help me identify the problem?" I would greatly appreciate your response

rdzman, 2023-05-01:
I would need more information in order to help understand your specific results. But I do have a few comments.

- The formulation assumes that inter-temporal constraints (ramp, etc) are enforced only on a "high probability" central path, with the contingency cases being excluded from these constraints, under the assumption that they are low probability events that require remedial action. So using such high probabilities for contingencies violates that assumption. For such high probabilities, you probably want to consider them as "scenarios" (like wind) rather than "contingencies".
- Are the negative generation and load dispatching you are seeing infeasible solutions? If so, have you checked that the success flag to be sure the solver is actually finding an optimal solution?

Mars0077, 2023-05-22:
Thank you for providing me with valuable information. considering the contigencies as scenarios suggestion was helpful and it worked. Unfortunately I have encountered another issue. Although there is only one transition matrix available, I need to define multiple contingencies with varying failure probabilities. would you offer a solution for this problem?

Mars0077, 2023-05-24:
I should have given an example to state my question more specifically. I have two contingencies and even more. they are as follows:
 (there are 6 time steps. The contingency 1 happens at time step 2 and and at time step 3 The contingency 2 happens at the same time with contingency 1) 

[F_BUS, T_BUS, BR_R, BR_X, BR_B, RATE_A, RATE_B,RATE_C, TAP, SHIFT, BR_STATUS, PF, ... 
    QF, PT, QT, MU_SF, MU_ST,ANGMIN, ANGMAX, MU_ANGMIN, MU_ANGMAX] = idx_brch;
contprofile1 = struct( ...
    'type', 'mpcData', ...
    'table', CT_TBRCH, ...
    'rows', 21, ...
    'col', BR_STATUS, ...
    'chgtype', CT_REP, ...
    'values', [] );
contprofile1.values(:,:, 1) = [1 1; 0 1; 0 1; 0 1; 0 1; 0 1];
profiles = getprofiles(contprofile1,profiles);

contprofile2 = struct( ...
    'type', 'mpcData', ...
    'table', CT_TBRCH, ...
    'rows', 7, ...
    'col', BR_STATUS, ...
    'chgtype', CT_REP, ...
    'values', [] );
contprofile2.values(:,:, 1) = [1 1; 1 1; 0 1; 0 1; 0 1; 0 1];
profiles = getprofiles(contprofile2,profiles);

% and the transition matrix

transmat = cell(1, nt);
lambda1=0.7;
% lambda2=0.4;
T = [lambda1; 1-lambda1];  
[transmat{:}] = deal(T * ones(1,2));
transmat{1} = [lambda1;1-lambda1];

Lambda1 is the failure probability of line 1 and lambda 2 is for the other line. I'm not sure how transission matrix is applied and how this code is functioning, But i guess it applies lambda 1 for both contingencies. But i need lambda 1 for contingency 1 and the lambda 2 for contingecy 2. 
I really appreciate your help

rdzman, 2023-05-24:
Ok, this helps ...

... so the question is how many discrete states (scenarios) do you want to model at each step. It sounds like you want something like the following:
- period 1 : 1 scenario (base case only, no contingencies)
- period 2 - 2 scenarios (base case w/probability (1-lambda1) and outage of line 1 with probability lambda1)
- period 3 - ?
- period 4 - back to a single base case state?

It's not clear to me what happens in period 3. Do you have 4 states (e.g. base case, line 1 out, line 2 out, both lines out)? Or 2 states (base case, both lines out)? Or 3 states (base case, line 1 out, both lines out)?

The bottom line is that you need to be able to specify all of the scenarios you want to include at each time period, and the transition probability matrices govern the probabilities of transition from each state in one period to each state in the next.

rdzman, 2023-05-25:
Ok, so it looks like you are trying to model all of the scenarios in the geometrically exploding scenario tree (2^5 = 32 scenarios in period 6). While MOST can model the probabilistic costs of these scenarios, the way it handles inter-temporal constraints such as ramping is based on an assumption of maintaining feasibility of worse-case transitions, which may or may not make sense for your problem.

But, if we order the scenarios in each period first by contingency, then by no outage first, then outage, then I think your transition probability matrices will look something like the following:
```
transmat{1} = 1;
transmat{2} = [ 1-lam1;
                lam1 ];
transmat{3} = [ 1-lam2  0;
                lam2    0;
                0       1-lam2;
                0       lam2 ];
transmat{4} = [ 1-lam3  0       0       0;
                lam3    0       0       0;
                0       1-lam3  0       0;
                0       lam3    0       0;
                0       0       1-lam3  0;
                0       0       lam3    0;
                0       0       0       1-lam3;
                0       0       0       lam3 ];
etc.
```

A different approach, which is probably simpler, and could potentially be closer to what you want, would be to define 32 separate 6-period problems, corresponding to the 32 possible system trajectories. Solve each as a deterministic problem, then probabilistically weight the results.

Mars0077, 2023-05-27:
I apologize for asking many questions and I also express my appreciation for your patience. I think the second approach would work well, But i wanted to try the transition matrix because it does the work with a lot less lines of codes. Now my question is that the matrix you wrote in the last comment seems to consider only one contingency at a time, While I need simultaneous contingencies at each time. I also have a question about How does the transition matrix determine which contingency to consider. I tried few experiment with the transition matrix and 2 profiles and I observed  that it was only applied to the last profile. I concluded that perhaps the transition matrix can only consider one contingency at a time and may not function with simultaneous contingencies. Please correct me if I'm mistaken, and if it is possible, could you explain how the transition matrix operates with simultaneous contingencies?

Mars0077, 2023-05-30:
I apologize for exceeding your patience limit. And I wanted to express gratitude for your advices and time and efforts you've put into helping me. I apologize again for any frustration or inconvenience My questions may have caused. I extend my sincerest wishes that you may always continue make a positive impact in the electrical engineering society.

rdzman, 2023-05-30:
The transition probability matrix controls transitions between different scenarios. The scenarios can be defined as generally as you like, that is any given scenario could have multiple contingencies active. In the example I gave above, the first period has two scenarios, one with line 1 in, one with it out, the second period has 4 scenarios, one with no contingencies, one with contingency on line 1, one with contingency on line 2, and one with contingency on both lines. Etc.

closed 2023-06-01 by Mars0077