MATPOWER/matpower #182: makePTDF: sparse matrix caused wrong results in makeLODF

determ1ne, 2023-05-12:
MATPOWER version: 7.1
MATLAB version: R2022b

In `makePTDF`, `Bbus` and `Bf` are sparse matrices returned by `makeBdc`.
https://github.com/MATPOWER/matpower/blob/67975794a660ea4cf5c7e59204a336c7c1fcbb30/lib/makePTDF.m#L123-L124

The variables are then used to calculate `dTheta` and PTDF(`H`).
https://github.com/MATPOWER/matpower/blob/67975794a660ea4cf5c7e59204a336c7c1fcbb30/lib/makePTDF.m#L155-L160

The sparse matrices calculation caused a small discrepancy in PTDF, and the error was enlarged in makeLODF by matrix division.
https://github.com/MATPOWER/matpower/blob/67975794a660ea4cf5c7e59204a336c7c1fcbb30/lib/makeLODF.m#L32

**Example (case 9):**

1. Get LODF for case9:

```matlab
mpc = case9;
H = makePTDF(mpc);
LODF = makeLODF(mpc.branch, H)
```

```matlab
LODF =

   -1.0000    0.0000    0.0000       NaN         0         0         0   -0.0000   -0.0000
   -1.5000   -1.0000   -1.0000       NaN   -1.0000   -1.0000         0   -1.0000   -1.0000
   -1.0000   -1.0000   -1.0000       NaN   -1.0000   -1.0000         0   -1.0000   -1.0000
         0         0         0       NaN         0         0         0         0         0
    0.5000   -1.0000   -1.0000       NaN   -1.0000   -1.0000         0   -1.0000   -1.0000
    0.5000   -1.0000   -1.0000       NaN   -1.0000   -1.0000         0   -1.0000   -1.0000
         0         0         0       NaN         0         0   -1.0000         0         0
    0.7500   -1.0000   -1.0000       NaN   -1.0000   -1.0000         0   -1.0000   -1.0000
    0.5000   -1.0000   -1.0000       NaN   -1.0000   -1.0000         0   -1.0000   -1.0000
```

2. Convert sparse matrices to dense before calculating `dTheta`:

`makePTDF.m`:

```matlab
Bbus = full(Bbus);
Bf = full(Bf);
```

Get LODF:

```matlab
mpc = case9;
H2 = makePTDF(mpc);
LODF = makeLODF(mpc.branch, H2)
```

```matlab
LODF =

       NaN         0    0.0000         0   -0.0000         0       NaN   -0.0000    0.0000
       Inf   -1.0000   -1.0000         0   -1.0000   -1.0000      -Inf   -1.0000   -1.0000
       Inf   -1.0000   -1.0000         0   -1.0000   -1.0000       Inf   -1.0000   -1.0000
       Inf   -0.0000    0.0000   -1.0000   -0.0000         0      -Inf         0    0.0000
      -Inf   -1.0000   -1.0000         0   -1.0000   -1.0000      -Inf   -1.0000   -1.0000
      -Inf   -1.0000   -1.0000         0   -1.0000   -1.0000      -Inf   -1.0000   -1.0000
       NaN         0         0         0         0         0       NaN         0         0
      -Inf   -1.0000   -1.0000         0   -1.0000   -1.0000       NaN   -1.0000   -1.0000
       Inf   -1.0000   -1.0000         0   -1.0000   -1.0000       NaN   -1.0000   -1.0000
```

3. The difference of PTDF between two calculations:

```matlab
>> H2-H

ans =

   1.0e-14 *

         0    0.1110    0.0888    0.0222    0.0444    0.0888    0.1110    0.1110    0.0777
         0    0.0611    0.0555    0.0306    0.0333    0.0555    0.0444    0.0555    0.0111
         0    0.0278    0.0777    0.0161    0.0361    0.0777    0.0555    0.0333    0.0111
         0    0.0214   -0.0222   -0.0011    0.0074   -0.0661         0         0         0
         0    0.0278   -0.0222    0.0079   -0.0028   -0.0222    0.0222    0.0222    0.0222
         0   -0.0722   -0.0389   -0.0067   -0.0472   -0.0389   -0.0444   -0.0888   -0.0222
         0    0.0444         0         0         0         0         0         0         0
         0   -0.0222   -0.0167   -0.0061         0   -0.0167   -0.0333   -0.0222   -0.0042
         0   -0.0555   -0.0722   -0.0315   -0.0472   -0.0722   -0.0666   -0.0555   -0.0444
```

rdzman, 2023-05-12:
LODF's for radially connected branches (such as branches 1, 4 and 7 in `case9`) are not well-defined, since they result in a network that is not fully connected.

In this case, the other columns of the LODF results are still well defined and match to within the expected numerical precision.

```
>> LODF(:, [2;3;5;6;8;9])-LODF2(:, [2;3;5;6;8;9])

ans =

   1.0e-14 *

   -0.3286    0.0889         0         0    0.0000    0.1778
         0    0.0888   -0.4663         0    0.3109    0.3331
    0.5107         0   -0.3775   -0.3220    0.2887    0.1998
   -0.0793    0.0260    0.0261         0         0    0.0029
    0.2442   -0.2554         0         0    0.4219    0.1110
    0.2442    0.2220   -0.7661         0    0.2887    0.3109
         0         0         0         0         0         0
    0.5329    0.0444   -0.1554         0         0    0.1554
    0.4441    0.0444   -0.1554    0.2109    0.0444         0
```

So, I don't think using sparse matrices internally in `makePTDF()` is a problem. Essentially, the small numerical differences in the resulting PTDFs result in widely varying values for quantities that are already ill-defined and therefore essentially meaningless.

determ1ne, 2023-05-13:
> LODF's for radially connected branches (such as branches 1, 4 and 7 in `case9`) are not well-defined, since they result in a network that is not fully connected.
>
> In this case, the other columns of [...]

Thank you for your reply! Is it necessary to add an additional check like tarjan in `makeLODF()` and set the corresponding lines to `NaN` ?

determ1ne, 2023-05-19:
> I agree that could be useful, at least as an option. There is a graph-traversal algorithm already implemented in [`connected_components()`](https://github.com/MATPOWER/matpower/blob/master/lib/connected_components.m) and used by [`case_info()`](https://github.com/MATPOWER/matpower/blob/master/lib/case_info.m), [`extract_islands()`](https://github.com/MATPOWER/matpower/blob/master/lib/extract_islands.m), and [`find_islands()`](https://github.com/MATPOWER/matpower/blob/master/lib/find_islands.m).
> 
> But, if I'm realistic, this is not likely to become a high priority for me. However, contributions are always welcome.

I tried to implement the option and found that `connected_components()` is not enough, because we need to find which branch will take apart the system (a "bridge" in graph theory, so I named them "bridge branches"). The proposed `find_bridges()` takes advantage of Tarjan's algorithm to find the bridges and it should be efficient.

A PR (#185) is submitted for this issue and I added an option to mask the corresponding matrix element to NaN. Tested on `case9` and modified `case16ci` .

**case9**:

```matlab
mpc = case9;
H = makePTDF(mpc);
LODF = makeLODF(mpc, H, 1);
```

```matlab
>> LODF = makeLODF(mpc, H, 1)

LODF =

       NaN    0.0000    0.0000       NaN         0         0       NaN   -0.0000   -0.0000
       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000
       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000
       NaN         0         0       NaN         0         0       NaN         0         0
       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000
       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000
       NaN         0         0       NaN         0         0       NaN         0         0
       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000
       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000       NaN   -1.0000   -1.0000
```

**case16ci modified**:

This system has three islands and one of them has a cycle.

```matlab
mpc = case16ci;
mpc.branch(14, :) = [5,7,0.00249570115476092,0.00249570115476092,0,0,0,0,0,0,1,-360,360];
H = makePTDF(mpc);
LODF = makeLODF(mpc, H, 1);
```

```matlab
>> LODF = makeLODF(mpc, H, 1)

LODF =

       NaN    0.0000    0.0000         0       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN         0         0   -1.0000
       NaN   -1.0000    1.0000    1.0000       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN   -1.0000         0   -0.5946
       NaN    1.0000   -1.0000   -1.0000       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN    1.0000         0   -0.4054
       NaN    1.0000   -1.0000   -1.0000       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN    1.0000         0   -0.4054
       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN         0         0
       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN    1.0000         0
       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN         0         0
       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN         0         0
       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN         0         0
       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN         0         0
       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN    0.5000         0
       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN   -1.0000   -1.0000
       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN   -2.0000   -4.0000
       NaN   -1.0000    1.0000    1.0000       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN   -1.0000         0   -0.5946
       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN   -1.0000         0
       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN       NaN         0   -1.0000
```

commit 58fe68fb (2023-05-30): Add tests and additional documentation for makeLODF(... mask_bridges) option.

Fix #182.

Ref: https://github.com/MATPOWER/matpower/pull/182
Ref: https://github.com/MATPOWER/matpower/pull/185
closed 2023-05-30 by rdzman