MATPOWER/matpower #184: Powerflow feasibility check

devsrivas, 2023-05-14:
Hi, I am looking to use matpower routines to do a feasibility check- given a set of generator and line outages, can the bus loads be supplied, or will load shedding/curtailment be needed.
Not having any gen/load schedules to begin with, I am using the OPF formulation with zeroed out costs - under the assumption that if all costs are same, the first feasible schedule will have the same overall cost, as all others.
Is this a fair assumption, or is there a better approach for this. Computation time is the main consideration.

Also, AC OPF is giving a warning 'Matrix is close to singular or badly scaled'.
Thanks.

rdzman, 2023-05-16:
Using the standard `runopf()` to check for feasibility is probably not the ideal, since it will fail to converge if infeasible, but there is still no guarantee that infeasibility was the cause for the failure to converge. I would suggest turning on OPF soft limits using [`toggle_softlims()`](https://github.com/MATPOWER/matpower/blob/master/lib/toggle_softlims.m), described in Section 7.6.4 of the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf). If load shedding is needed, it should show up as some constraint violation.

However, the bigger issue, which can't be solved by soft limits, is that of islanding and disconnected network components. That is, if there is a single line that connects a portion of the network to the rest of the system, an outage of that line will cause numerical problems if you try to run the OPF. That may be the cause of the error you are seeing.

The only way I know to check for that is graph traversal. For that you might try [`find_islands()`](https://github.com/MATPOWER/matpower/blob/master/lib/find_islands.m) and/or [`extract_islands()`](https://github.com/MATPOWER/matpower/blob/master/lib/extract_islands.m). With an islanded system, you simply need a reference bus to be assigned in each island and then you could run the OPF with soft limits on each island to see if the resulting system is still feasible.

devsrivas, 2023-05-21:
Thanks for your response. Would a DC OPF also be subject to the same issue?

devsrivas, 2023-05-21:
I did an experiment on this. Taking the 24 bus IEEE-RTS case I created an island by opening several branches.
While in AC OPF, it failed stating the singularity of the matrix, in DC OPF it failed for lack of a feasible solution,
for the same network state.

rdzman, 2023-05-22:
Yes, both AC and DC OPF are subject to the same issue when line outages create islands. In order to solve the islanded system, even with soft-constraints, you will need to ensure that there is a voltage reference bus in each island.

devsrivas, 2023-05-31:
Another experiment: Ran this for a thousand monte carlo simulations, filtered out cases with more than one islands, and only ran AC OPF for a single island. Saw the same error for several runs:
Warning: Matrix is close to singular or badly scaled. Results may be inaccurate. RCOND =  1.454522e-22.

rdzman, 2023-05-31:
If the OPF converged, you can ignore the warnings. If not, then you may need to try a different solver like IPOPT, which is probably a bit more robust than the included default MIPS solver.

closed 2024-04-16 by rdzman