MATPOWER/matpower #187: Finding isolated buses for contingency cases

thomsonian2023, 2023-06-09:
I am working with PSSE RAW files and I have a few contingencies defined with line and transformer outages. Is there a way that I can use matpower to figure out what buses become isolated under those contingencies i.e. when all those defined lines and transformers go on outage at same time?

rdzman, 2023-06-09:
If you have a MATPOWER case of the topology of interest, you can use [`find_islands()`](https://github.com/MATPOWER/matpower/blob/master/lib/find_islands.m) or [`case_info()`](https://github.com/MATPOWER/matpower/blob/master/lib/case_info.m) to find the isolated buses. There is also an [`extract_islands()`](https://github.com/MATPOWER/matpower/blob/master/lib/extract_islands.m), in case you would find that useful.

thomsonian2023, 2023-06-09:
In the documentation of "find_islands" it says about the variable "isolated": "indices of isolated buses that have no connecting branches." So, how do I get the master list of buses to which the indices "isolated" can be applied?

rdzman, 2023-06-13:
The values returned by [`find_islands()`](https://github.com/MATPOWER/matpower/blob/master/lib/find_islands.m) correspond to row indices into the `mpc.bus` matrix. They may not be the same as the external bus numbers in the `BUS_I` column of `mpc.bus`.

closed 2024-04-16 by rdzman