MATPOWER/matpower #191: after installing gurobi, how to let matpower know it?

yanda1234, 2023-06-30:
Hi Dr. Zimmerman,

Sorry to disturb you before holiday. For some reason, I need to install gurobi again, but after that, I found matpower cannot recognize it.
I believe gurobi is installed and the link between matlab and gurobi is also installed, since I tested a gurobi case from matlab.
However, the problem is matpower cannot call gurobi. I tried mpoption('gurobi.method',1) and mpver. And the first command brought back me an error message, while the second one got me gurobi is not installed. Do you know how to fix it? Kindly advise.

Many thanks.

Best,
Yanda

rdzman, 2023-06-30:
MATPOWER checks once for the presence of external solvers, then caches the result to make future checks faster. You can clear that cache by doing ...
```
have_feature('all', 'clear_cache')
```
... or simply restart you MATLAB session. Then it should find it.

yanda1234, 2023-06-30:
Hi Dr. Zimmerman,

This function doesn't work for me. 
[image]
[image]
[image]

And I restarted matlab, but matpower still cannot recognize gurobi. Do you have any thoughts? 

Best,
Yanda

rdzman, 2023-06-30:
What version of MATPOWER are you using?
And what is the output when you type: `which gurobi` ?

yanda1234, 2023-06-30:
[image]
Matpower version is 7.0

rdzman, 2023-06-30:
The fact that it can't find `have_feature()` means that either you are using a version of MATPOWER before 7.1, or it is not properly installed.

But if Gurobi is in the MATLAB path after restart, it should find it anyway.

yanda1234, 2023-06-30:
Actually, I need to type gurobi_setup each time right now. Do you mean that I can either use matpower 7.1 or ensure gurobi is in matlab path?

rdzman, 2023-06-30:
Hmm, I'm not aware of any issues with MATPOWER 7.0 and Gurobi 9.5.2.
What does `exist('gurobi', 'file')` return?
And what happens if you try the following?
```
model = struct( ...
    'A', sparse(1), ...
    'rhs', 1, ...
    'sense', '=', ...
    'vtype', 'C', ...
    'obj', 1, ...
    'modelsense', 'min' ...
);
params = struct( ...
    'outputflag', 0 ...
);
result = gurobi(model, params);
```

rdzman, 2023-06-30:
Yes, Gurobi must be in the MATLAB path for MATPOWER to find it ... regardless of the version of MATPOWER.

yanda1234, 2023-06-30:
If I add path each time, instead of saving there permanently, is it ok?

yanda1234, 2023-06-30:
[image]

rdzman, 2023-06-30:
As long as you add it to your path **before** MATPOWER tries to check for its availability you should be fine. If MATPOWER checks first, then it will not find it and will cache that result and not check again.

yanda1234, 2023-06-30:
I see. Is there any problem for the model, params and result?

rdzman, 2023-06-30:
No, I was just hoping it would produce an error that would help me figure out the problem. But you must have had Gurobi in your path that time, so no error.

rdzman, 2023-06-30:
But it's not a bad idea to update to the latest MATPOWER.

yanda1234, 2023-06-30:
Thanks, Dr. Zimmerman. Let me try to ask administrator save gurobi path permanently. If it still doesn’t work, I will update matpower to 7.1. Hopefully, it will be ok by then.

yanda1234, 2023-06-30:
Hi Dr. Zimmerman,

After I cleared cache, the mpver works, which means matpower can recognize gurobi, right?
[image]
However, mpoption still doesn't work. Do you have any thoughts?
[image]

Best,
Yanda

rdzman, 2023-06-30:
I don't understand how that can happen. Can you start a new MATLAB session, make sure Gurobi is in the path **before** using any MATPOWER commands, then try `mpver` and `mpoption('gurobi.method', 1)`?

Both should work. If not, maybe something got corrupted with your MATPOWER install and you need to reinstall (or better yet, upgrade) MATPOWER.

yanda1234, 2023-06-30:
It is interesting. It works this time, although I still don't know why... Thanks a lot.

closed 2023-06-30 by yanda1234