MATPOWER/matpower #192: Most Multi-Period Simulation

ragt2312, 2023-07-11:
Hello Mr. Zimmerman

I am having a problem when running a multi-period simulation with MOST, basically, I get different results when I raise the number of simulation periods. For example, if I simulate 2 periods I get the results that I expect for those periods, but when I simulate 3 periods the results in those first 2 periods are different and again different when simulating 4, 5 etc. Shouldn't the first two periods be independent of any future ones?

Thank you

rdzman, 2023-07-11:
No. In general, earlier periods should **not** be independent of future ones. If that were the case, there would be no reason to optimize them together and you could simply solve single-period problems sequentially.

For example, ramping constraints can result in a situation where a slightly more expensive dispatch in period 2 will allow for a much lower cost dispatch in period 3. E.g. pre-ramping a slow cheap unit needed to meet period 3 demand. If you optimize over periods 1 and 2 only, you will not be able to achieve this lower cost solution for the 3-period problem.

Hope this helps.

ragt2312, 2023-07-11:
Thank you for the fast response, I understand what you are saying, but in my case I gave all the generators very high ramp rates to make some tests, doing this shouldn't I get the same results as multiple single period simulations solved sequentially?

Kind regards

rdzman, 2023-07-11:
If you get rid of **all** inter-temporal costs and constraints **and** there is a unique solution to the single-period problems, then yes.

Intertemporal costs and constraints in MOST include storage, hard ramp limits, load-following ramp reserve costs and limits, ramp wear-and-tear costs, and for unit commitment min-up, min-down constraints, startup and shutdown costs.

If you get rid of all of those, at least the cost of the dispatches in each period should not change. The dispatches themselves might change if the solution space is flat.

ragt2312, 2023-07-11:
Thank you for your answers Mr. Zimmerman.

Kind Regards

closed 2024-04-16 by rdzman