MATPOWER/matpower #193: Setting of PositiveActiveReserveQuantity

yadong-zhang, 2023-07-20:
When using MOST to solve multiperiod UC problem, I am bit confused about the setting of "PositiveActiveReserveQuantity" in the xgd_table file. It means the maximum ramp up reserve quantity, which should equal (max_gen_capacity - current_gen_capacity). But aren't max_gen_capacity and current_gen_capacity already set in the .mpc file? Why should we do it again? Will these two be in conflict with each other?

Thanks a lot for your help!

rdzman, 2023-07-20:
`PositiveActiveReserveQuantity` is the maximum upward deviation of the dispatch, in any base or contingency case, from the active power contract (not any actual dispatch point). The active power contract is also an optimization variable.

So the `PositiveActiveReserveQuantity + NegativeActiveReserveQuantity` defines the magnitude of the range of possible dispatches across all scenarios/contingencies in terms of offered range of dispatch. The dispatches are also restricted by physical ramp rates (i.e. `mpc.gen(:, RAMP_30`).

yadong-zhang, 2023-07-20:
> `PositiveActiveReserveQuantity` is the maximum upward deviation of the dispatch, in any base or contingency case, from the active power contract (not any actual dispatch point). The active power [...]

Thanks for the clarification!

closed 2024-04-16 by rdzman