MATPOWER/matpower #197: How to get particular jacobians and sensitivity matrix

akhtar3100, 2023-07-27:
I have two questions:
1. if I use makeJac(mpc) I can get the jacobian matrix. Is it possible to get for example J4=dQ/dVm directly with some command rather than finding exact location/indices of these submatrices. 
2. My second question is related to first one, how to do sensitivity analysis using matpower. If we need to find for example dploss/dq or dqloss/dq how we can find it using matpower?

rdzman, 2023-07-27:
These are good questions.

1. If you are working with a system that has been converted to internal numbering via `ext2int()` and you have the complex bus voltage vector `V`, you can use `dSbus_dV()` to compute the parts of the Jacobian separately as ...
```
[Ybus, Yf, Yt] = makeYbus(mpc);
[dSbus_dVa, dSbus_dVm] = dSbus_dV(Ybus, V);
dQ_dVm = imag(dSbus_dVm);
```

2. You could use `dSbr_dV()` to get the sensitives of line flows (losses equal sum of Sf and St) w.r.t the voltages, but that's not w.r.t. q.

akhtar3100, 2023-07-28:
Ok.  What is the use of ext2int function? what if I dont use it. I see no difference between using it or not? One more thing, where i can find details of the functions like dSbus_dv() and dSbr_dv()?

akhtar3100, 2023-08-01:
@rdzmanWhat is the use of ext2int function? what if I don't use it. I see no difference between using it or not?
Here we are calculating dQ/dVm, if I want to calculated dVm/dQ will I need to take inverse of the matrix?
One more thing, where i can find details of the functions like dSbus_dv() and dSbr_dv()? do you have function for dVm_dQ?
Regarding losses, if I want to calculate losses caused by reactive power flow and loss sensitivity to reactive power, how should I calculate it.

rdzman, 2023-08-03:
_(Sorry for the delay in responding. I was out for several days.)_

1. It removes offline equipment (generators, branches) and renumbers buses consecutively. If all buses are already numbered consecutively and there is no offline equipment, it will not make a difference.
2. Yes.
3. You can use MATLAB's `help` command for basic help. The details of the formulae used for the calculations can be found in [“AC Power Flows, Generalized OPF Costs and their Derivatives using Complex Matrix Notation”](https://matpower.org/docs/TN2-OPF-Derivatives.pdf).
4. Not sure what you mean by "losses caused by reactive power flow"? The total active power loss in the system is a function of all of the variables (bus voltages and bus complex power injections. In any case, you can use [`get_losses()`](https://github.com/MATPOWER/matpower/blob/master/lib/get_losses.m) to compute the losses for a given operating point, along with sensitivities w.r.t bus voltages. However, to get sensitivities w.r.t reactive power injections, you will need to make some assumptions about how other variables are constrained when Q is perturbed. I have a function that computes AC shift factors at a given operating point, with certain assumptions about slacks and fixed voltages that I've attached here. As mentioned previously, the sensitivity of losses in a branch should simply be the sum of the sensitivities of flows at the two ends. You may find [makePTDFac.m.txt](https://github.com/MATPOWER/matpower/files/12252871/makePTDFac.m.txt) useful as is, or at least find the code illuminating for your application _(remember to strip the `.txt` extension)_.

closed 2023-09-22 by akhtar3100