MATPOWER/matpower #198: How to perform powerflow with generators with specefic Pg and Qg

akhtar3100, 2023-08-01:
I have noticed that when we runpf in matpower it only considers Pg of generators and Qg of generators is calculated, That is also what basic powerflow does. But is it possible to runpf with specefic values of generators Pg and Qg given, I want powerflow to consider that for generators active and reactive powers both are given and then run powerflow, how can I do that?

rdzman, 2023-08-03:
For a power flow, if you want to specify both P and Q for a generator, rather than P and |V|, simply change the bus type _(`BUS_TYPE` column in the `bus` matrix)_ from 2 _(PV)_ to 1 _(PQ)_.

Dispatchable loads are only for the OPF.

akhtar3100, 2023-08-04:
Ok. thanks for reply. In this case I will keep bus pd and qd as zero and generator pg and qg will be negative values, right?

rdzman, 2023-08-04:
Generator QG can be positive or negative, but will scale with PD as it is dispatched, maintaining a constant power factor.

closed 2023-09-22 by akhtar3100