MATPOWER/matpower #200: DCOPF not working on some pglib-opf cases

dobakhshari, 2023-08-18:
Hello
I was running rundcopf on pglib-opf[](https://github.com/power-grid-lib/pglib-opf)  cases, which I found give error of did not converge with success being 0.
This happened for most cases of more than 30 buses if I set a single line flow limit below that of the optimal case without any flow limits activated. For example for the most loaded line in pglib_opf_case60_c.m if the flow limit changes from 700 MW to 650MW. The original case,by the wsy, faces no line flow limit.
 For cases with +1000 buses the original case does not converge at all, e.g., pglib_opf_case9241_pegase.m
Thank you

>

ccoffrin, 2023-08-18:
It has been a long time since I have tested the [pglib-opf](https://github.com/power-grid-lib/pglib-opf) cases in MATPOWER, but I might be able to provide some insights on this inquiry. The last time I did this test was back in 2017, so quite some time ago with old versions of both PGLib and MATPOWER.  Table 1 in [this paper](https://arxiv.org/pdf/1711.01728.pdf) shows the comparison.  At that time I did observe there are some cases in pglib where there is no feasible solution in the context of DC OPF (the SAD cases in particular). In these cases I would not expect MATPOWER (or any other software) to find a solution.

In the class of TYP cases, which you seem to be referring to here, I do belive MATPOWER and other software tools should find a DC OPF solution.  So the issue you observe is unexpected to me. @rdzman's suggestion for debugging sounds like to best next step to me as well.

dobakhshari, 2023-08-19:
@ccoffrin  Thanks for your comment. I took a look at Table 1 and all typical cases seem good. 
@rdzman   I believe the simple case of[ pglib_opf_case60_c.m](https://github.com/power-grid-lib/pglib-opf/blob/master/pglib_opf_case60_c.m) with line 26-27 flow set from 700 to 650 MW should not be infeasible and the problem must be the solver. 
I tried MATLAB OPT toolbox, as I believe linprog should handle this light case. Based on Table 6.2 of the manual I did the following.
`mpopt = mpoption('opf.dc.solver','OT');` 
`results = rundcopf('pglib_opf_case60_c',mpopt)` 
And it still does not work.
Would you please give a simple code to use linprog as the manual seems silent on it?

rdzman, 2023-08-21:
I am not able to duplicate the issue you are seeing. When I run the commands you listed above it solves just fine. With the original case ``pglib_opf_case60_c.m``, it results in a flow of 694.13 MW in branch 21 (bus 26-bus 27). If I modify the ``RATE_A`` for that branch from 700 MW to 650 MW, `linprog` still solves it just fine for me, with a resulting flow of 650 MW in that line.

Is it possible that you are using a version of the case that has other modifications? Maybe try a fresh download of the case file just to double-check.

dobakhshari, 2023-08-25:
My setof codes still fails to solve the DCOPF. However, I found that the default algorithm used in linprog is interior point; when I changed it to simplex everything got right.

closed 2023-08-25 by dobakhshari

rdzman, 2023-08-25:
👍