MATPOWER/matpower #205: Matpower 3.2, issue with network with more than 1 swing/slack bus

JasonMic97, 2023-09-30:
Hi team MATPOWER
My name is Jason Michael and I'm a Master's student in Electrical Engineering. I am currently finishing my degree and working on Optimal Power Flow. I'm using version 7.1 and for a specific case of Optimal Reactive Power Dispatch, I'm also using version 3.2. I am writing to you to ask for some clarification. I'm working on a network with 3 slack buses and the power flow process with version 7.1 works perfectly, but with version 3.2, I'm encountering the following problem:

Matrix dimensions must agree.

Error in pfsoln (line 39)
refgen = find(gbus == ref);             %% which is(are) the reference gen(s)?

Error in runpf (line 146)
        [bus, gen, branch] = pfsoln(baseMVA, bus, gen, branch, Ybus, Yf, Yt, V, ref, pv, pq);
 
So, with all due respect, I would like you to help me with a solution to this problem. 
Very respectfully

JasonMic97, 2023-09-30:
Well guys, happily the AI chatgbt helped me, now it is working. this is the solution :
%% Update Pg for swing bus
for k = 1:length(ref)
    refgen = find(gbus == ref(k));  %% Determine which generators are connected to the reference bus
    for j = 1:length(refgen)       %% Iterate through generators at the reference bus
        gen(on(refgen(j)), PG) = real(Sg(refgen(j))) * baseMVA + bus(ref(k), PD); %% Compute generator real power output
    end
end

rdzman, 2023-10-02:
Ok.

I'm just curious ... what is the reason behind using MATPOWER 3.2 (released ~16 years ago)? I'm a little surprised that it even runs on modern versions of MATLAB.

JasonMic97, 2023-10-02:
Hi Professor.

My reason for using this version is that I use matlab code from
Optimal Reactive Power Dispatch (found on Matlab File Exchange); This
code works with version 3.2, I remodified it to use it with version
7.1, but there were some variable declarations that didn't match and
so I opted for 3.2 for this ORPD work. Otherwise finally I found the
solution to the multi swing bus problem thanks to ChatGBT. I am using
Matlab version 2020ra

closed 2024-04-16 by rdzman