MATPOWER/matpower #206: Matpower install issue

akhtar3100, 2023-10-13:
Hi! I have installed matpower in matlab 2023b (linprog and other problems solved) but somehow whenever I restart my computer I need to reinstall the matpower. I have changed the path of matlab directory according to matpower but still facing the same problem.

rdzman, 2023-10-13:
Did you select option 3 (to save the updated path after modifying it) when running `install_matpower`?

With option 3, the installer just runs an `addpath()` command and then calls MATLAB's `savepath()` function to save those changes to the `pathdef.m` file that MATLAB uses to define the default path when it starts up. It sounds like, in your case, either ...

1. You did not use option 3, so the updated path is never saved.
2. For some reason, MATLAB is unable to successfully execute `savepath()`, maybe due to permissions issues.
3. Rebooting your computer causes the `pathdef.m` file to get restored to some previous state (e.g. in a university computer lab where machines are restored  to some known state on reboot).

If it's #1, the solution is easy. Just install using option 3. For the other two, my suggestion would be to install MATPOWER using option 1, and save the `addpath()` command to a file. You can then call that file from a `startup.m` file in your default MATLAB working directory (or copy the `addpath()` command directly to `startup.m`) so that it gets executed on startup each time. This is actually the approach I use since it allows me to use different `startup.m` files if I want to run older versions of MATPOWER.

In case it helps for debugging, you can use MATLAB's `path` command to see what directories are included in the MATLAB path for the current session. After running `install_matpower` with option 2 or 3, the MATPOWER directories should be at the end of the list. You can also do `type pathdef` to see the contents of the `pathdef.m` file showing what will be included in the default path when MATLAB is restarted.

akhtar3100, 2023-10-16:
Hi,
I was using option 3 as you mentioned in bullet point 1 but still the path is probably not being saved. I think the reason is same as you said in point 3 that laptop is restored to some previous state. I have now installed using option 1, but I am not sure in which path i should copy addpath() file to. There are multiple startup.m files in my MATLAB directory in C.

akhtar3100, 2023-10-16:
I tried to add path file but when I run type pathdef it still does not show the directory where matpower is installed. is there any command to add the path directly to stratup.m

akhtar3100, 2023-11-06:
Thank you. I resolved the issue.

closed 2023-11-06 by akhtar3100