MATPOWER/matpower #207: matpower7 in matlab2020a test error

NoIdeaOfName66, 2023-11-13:
[image]

I was able to use matpower successfully before, but I don't know why something suddenly went wrong, so I uninstalled and reinstalled, but the above problems occurred during the reinstallation

NoIdeaOfName66, 2023-11-13:
I delet Mosek solver，then：
Would you like to run the MATPOWER tests now? [Y or N] (default = N) : Y
t_test_fcns.............ok
t_have_feature..........ok
t_mplinsolve............ok (6 of 66 skipped)
t_mips..................ok
t_mips_pardiso..........ok (60 of 60 skipped)
t_qps_mips..............ok
t_have_fcn..............ok
t_nested_struct_copy....ok
t_nleqs_master..........ok (30 of 150 skipped)
t_qps_master............not ok
	#####  Ran 432 of 432 tests: 214 passed, 2 failed, 216 skipped
t_miqps_master..........not ok
	#####  Ran 288 of 288 tests: 156 passed, 2 failed, 130 skipped
t_nlps_master...........ok (108 of 540 skipped)
t_opt_model.............ok
t_om_solve_leqs.........ok
t_om_solve_nleqs........ok (36 of 194 skipped)
t_om_solve_qps..........not ok
	#####  Ran 387 of 387 tests: 205 passed, 2 failed, 180 skipped
t_om_solve_miqps........ok (37 of 118 skipped)
t_om_solve_nlps.........ok (94 of 504 skipped)
t_qps_matpower..........not ok
	#####  Ran 396 of 396 tests: 214 passed, 2 failed, 180 skipped
t_miqps_matpower........not ok
	#####  Ran 288 of 288 tests: 156 passed, 2 failed, 130 skipped
t_feval_w_path..........ok
t_mpoption..............ok
t_loadcase..............ok
t_ext2int2ext...........ok
t_jacobian..............ok
t_hessian...............ok
t_margcost..............ok
t_totcost...............ok
t_modcost...............ok
t_hasPQcap..............ok
t_pf_ac.................ok
t_pf_dc.................ok
t_pf_radial.............ok
t_cpf...................ok
t_islands...............ok
t_opf_model.............ok
t_opf_model_legacy......ok
t_opf_default...........ok
t_opf_fmincon...........ok (62 of 872 skipped)
t_opf_mips..............ok (280 of 1308 skipped)
t_opf_dc_default........ok
t_opf_dc_cplex..........ok
t_opf_dc_gurobi.........ok
t_opf_dc_mips...........ok
t_opf_dc_mips_sc........ok
t_opf_dc_ot.............ok
t_testglobalopt.........ok
t_opf_userfcns..........ok
t_opf_softlims..........ok
t_runopf_w_res..........ok
t_dcline................ok
t_get_losses............ok
t_load2disp.............ok
t_makePTDF..............ok
t_makeLODF..............ok
t_printpf...............ok
t_vdep_load.............ok
t_total_load............ok
t_scale_load............ok
t_apply_changes.........ok
t_psse..................ok
t_off2case..............ok
t_auction_mips..........ok
t_runmarket.............ok
Ran 14253 of 14253 tests: 12694 passed, 10 failed, 1549 skipped
Elapsed time 70.21 seconds.

NoIdeaOfName66, 2023-11-13:
I install matpower8 replace matpower7 in matlab2020a, then:
Would you like to run the MATPOWER tests now? [Y or N] (default = N) : Y
t_test_fcns...............ok
t_have_feature............ok
t_mplinsolve..............ok (8 of 180 skipped)
t_mips....................ok
t_mips_pardiso............ok (60 of 60 skipped)
t_qps_mips................ok
t_have_fcn................ok
t_nested_struct_copy......ok
t_nleqs_master............ok (30 of 150 skipped)
t_pnes_master.............ok
t_qps_master..............ok (216 of 432 skipped)
t_miqps_master............ok (130 of 288 skipped)
t_nlps_master.............ok (108 of 540 skipped)
t_opt_model...............ok
t_om_solve_leqs...........ok
t_om_solve_nleqs..........ok (36 of 194 skipped)
t_om_solve_pne............ok
t_om_solve_qps............ok (180 of 387 skipped)
t_om_solve_miqps..........ok (37 of 118 skipped)
t_om_solve_nlps...........ok (94 of 504 skipped)
t_qps_matpower............ok (180 of 396 skipped)
t_miqps_matpower..........ok (130 of 288 skipped)
t_feval_w_path............ok
t_mpoption................ok
t_loadcase................ok
t_ext2int2ext.............ok
t_jacobian................ok
t_hessian.................ok
t_margcost................ok
t_totcost.................ok
t_modcost.................ok
t_hasPQcap................ok
t_mp_mapped_array.........ok
t_mp_table................ok (12 of 332 skipped)
t_mp_data_model...........ok
t_dmc_element.............ok
t_mp_dm_converter_mpc2....ok
t_nm_element..............ok
t_port_inj_current_acc....ok
t_port_inj_current_acp....ok
t_port_inj_power_acc......ok
t_port_inj_power_acp......ok
t_node_test...............ok
t_run_mp..................ok
t_run_mp_3p...............ok
t_pretty_print............ok
t_mpxt_reserves...........ok
t_pf_ac...................ok
t_pf_dc...................ok
t_pf_radial...............ok
t_cpf.....................ok (97 of 415 skipped)
t_islands.................ok
t_opf_model...............ok
t_opf_default.............ok
t_opf_fmincon.............ok (62 of 896 skipped)
t_opf_mips................ok (286 of 1344 skipped)
t_opf_dc_default..........ok
t_opf_dc_cplex............ok
t_opf_dc_gurobi...........ok
t_opf_dc_mips.............ok
t_opf_dc_mips_sc..........ok
t_opf_dc_ot...............ok
t_opf_userfcns............ok
t_opf_softlims............ok
t_runopf_w_res............ok
t_dcline..................ok
t_get_losses..............ok
t_load2disp...............ok
t_makePTDF................ok
t_makeLODF................ok
t_printpf.................ok
t_vdep_load...............ok
t_total_load..............ok
t_scale_load..............ok
t_apply_changes...........ok
t_psse....................ok
All tests successful (15520 passed, 1666 skipped of 17186)
Elapsed time 125.79 seconds.

maybe the question is solved ...

NoIdeaOfName66, 2023-11-13:
Now I have a new question: when I install matpower8 in matlab2023a, matlab crashes...

rdzman, 2023-11-13:
Regarding the first error, no need to completely delete MOSEK. By default, MOSEK includes replacements for a number of the MATLAB Optimization Toolbox functions, in this case `linprog()`, but they are not 100% compatible which causes problems for MATPOWER.

The solution is to remove these replacement functions from the MATLAB path. My suggestion is to type `which linprog` at the MATLAB prompt to find the path to the `lingprog()` function installed by MOSEK. Open the folder containing that file, create a sub-folder called something like `ot-replacments` and move the following files into that sub-folder to disable them (remove them from the MATLAB path):
```
intlinprog.m
linprog.m
lsqlin.m
lsqnonneg.m
quadprog.m
```
Then, restart MATLAB and try running the tests again. This is how I handle the MOSEK installation and it gives me the ability to use both MOSEK and the Optimization Toolbox, not have MOSEK disable and take over for the Optimization Toolbox (which I don’t want).

The second error, I believe, is related to an incompatible change made in the MATLAB Optimization Toolbox 9.5 in R2023a. It was fixed in the master branch of [MP-Opt-Model][1], but that has not yet been included in a new release of MATPOWER. You can either grab the latest [MATPOWER][2] from GitHub, which includes that update, or grab the latest [MP-Opt-Model][1] to replace the `mp-opt-model` subdirectory in your current MATPOWER 8.0b1 installation. Either one should allow everything to work fine on the latest MATLAB versions.

I know ... a new release of MATPOWER with this fix is long overdue. Sorry for the trouble.

[1]: https://github.com/MATPOWER/mp-opt-model
[2]: https://github.com/MATPOWER/matpower

closed 2024-04-16 by rdzman