MATPOWER/matpower #211: radial_pf() assumes bus 1 is the slack

rdzman, 2023-11-27:
As can be seen here ... 

https://github.com/MATPOWER/matpower/blob/fbe9cac672f121997f4b7e7e12e9408744d3646c/lib/radial_pf.m#L55-L58

As a temporary protection, until this incorrect assumption is fixed in the code, we need to put in an `assert()` that points to this issue for the case where the assumption of bus 1 being the slack is not true.

todorovski-m, 2023-12-02:
The issue isn't a bug. Within `radial_pf`  an initial call to `order_radial` handles the consecutive ordering of buses and assigns the number 1 to the slack bus. In `case18`  the numbering isn't consecutive and the original slack bus number isn't 1. This network, included in the test suite, is solved correctly.

rdzman, 2023-12-05:
👍 @todorovski-m, thanks for clarifying this. I didn't remember that about `order_radial`.

closed 2023-12-05 by rdzman