MATPOWER/matpower #216: how to model transformer and tap change of transformer in matpower

akhtar3100, 2023-12-21:
I have these questions.

1. Right now normally transformers are not included in casedata or mpc files. How can I add transformers in my networks? I want to have HV/MV transformer.
2. if I have a transformer in my mpc file, how can I change its taps and then runpf.

rdzman, 2024-01-02:
Transformers are included in the `branch` matrix. See Section 3.2 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf) for details of the model.

To change the taps, modify value in the `TAP` column.

akhtar3100, 2024-01-08:
Ok. Thanks. But one question, Are there any case files which include transformers in them.

rdzman, 2024-01-08:
I don't know without checking. There are essentially two things to look for.

First, any branch that has a non-zero `TAP` value is a transformer. Second, you can look for a case with buses that have different `BASE_KV` values. Any branch that connects buses with different `BASE_KV` would be a transformer.

akhtar3100, 2024-01-17:
Ok. I found that case14.m have 3 transformers between bus 4 &7, 4&9 and 5&6, now in the ratio columns the ratio is given 0.978, 0.969 etc. What does this value means? Is it TAP ratio? And one more thing, there could be transformer without tap setting. how to differentiate between a transformer with TAP and without TAP.

rdzman, 2024-01-17:
See Section 3.2 and Table B-3 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf). `TAP` is the off-nominal turns ratio. I assume by "without TAP" that you mean a transformer at it's nominal turns ratio, in which case `TAP` would be 1.

akhtar3100, 2024-01-19:
ok. Thanks. It means Tap=1 means nominal ratio and any other ration for example 0.98 or 1.02 would be off nominal ratio.

rdzman, 2024-01-19:
Exactly.

closed 2024-03-21 by akhtar3100