MATPOWER/matpower #218: can DCOPF be solved on a case containing DC lines

yanda1234, 2023-12-29:
Hi Dr. Zimmerman,

I attempted to address a DCOPF on case "t_case9_dcline" containing DC lines but failed in the process. Would it be possible to seek your guidance on whether such an undertaking is feasible within the capabilities of Matpower?
Kindly advise!

MATPOWER Version 7.0, 20-Jun-2019 -- DC Optimal Power Flow
Error using linprog
LINPROG(f,A,b,Aeq,beq,LB,UB,X0,OPTIONS) does not accept X0. Use LINPROG(f,A,b,Aeq,beq,LB,UB,OPTIONS) instead.

Error in qps_ot (line 267)
           linprog(c, Ai, bi, Ae, be, xmin, xmax,x0, ot_opt);

Error in qps_matpower (line 258)
            qps_ot(H, c, A, l, u, xmin, xmax, x0, opt);

Error in dcopf_solver (line 106)
[x, f, info, output, lambda] = qps_matpower(HH, CC, A, l, u, xmin, xmax, x0, opt);

Error in opf_execute (line 48)
  [results, success, raw] = dcopf_solver(om, mpopt);

Error in opf (line 232)
    [results, success, raw] = opf_execute(om, mpopt);

Error in dcopf (line 19)
[varargout{1:nargout}] = opf(mpc, mpopt);

Best,
Yanda

yanda1234, 2023-12-30:
Hi Dr. Zimmerman,

I build a case with some dclines and dcline costs. However, after running DCOPF, I found those dc lines were not really used since the branch flows of the branches connected their terminal buses didn't change at all. Do you agree? Kindly let me know your thoughts.

Best,
Yanda

yanda1234, 2023-12-30:
function mpc = prepared_case
%PREPARED_CASE

%% MATPOWER Case Format : Version 2
mpc.version = '2';

%%-----  Power Flow Data  -----%%
%% system MVA base
mpc.baseMVA = 100;

%% bus data
%	bus_i	type	Pd	Qd	Gs	Bs	area	Vm	Va	baseKV	zone	Vmax	Vmin
mpc.bus = [
	1000001	2	51	27	0	0	1	0.955	10.67	138	1	1.06	0.94;
	1000002	1	20	9	0	0	1	0.971	11.22	138	1	1.06	0.94;
	1000003	1	39	10	0	0	1	0.968	11.56	138	1	1.06	0.94;
	1000004	2	39	12	0	0	1	0.998	15.28	138	1	1.06	0.94;
	1000005	1	0	0	0	-40	1	1.002	15.73	138	1	1.06	0.94;
	1000006	2	52	22	0	0	1	0.99	13	138	1	1.06	0.94;
	1000007	1	19	2	0	0	1	0.989	12.56	138	1	1.06	0.94;
	1000008	2	28	0	0	0	1	1.015	20.77	345	1	1.06	0.94;
	1000009	1	0	0	0	0	1	1.043	28.02	345	1	1.06	0.94;
	1000010	2	0	0	0	0	1	1.05	35.61	345	1	1.06	0.94;
	1000011	1	70	23	0	0	1	0.985	12.72	138	1	1.06	0.94;
	1000012	2	47	10	0	0	1	0.99	12.2	138	1	1.06	0.94;
	1000013	1	34	16	0	0	1	0.968	11.35	138	1	1.06	0.94;
	1000014	1	14	1	0	0	1	0.984	11.5	138	1	1.06	0.94;
	1000015	2	90	30	0	0	1	0.97	11.23	138	1	1.06	0.94;
	1000016	1	25	10	0	0	1	0.984	11.91	138	1	1.06	0.94;
	1000017	1	11	3	0	0	1	0.995	13.74	138	1	1.06	0.94;
	1000018	2	60	34	0	0	1	0.973	11.53	138	1	1.06	0.94;
	1000019	2	45	25	0	0	1	0.963	11.05	138	1	1.06	0.94;
	1000020	1	18	3	0	0	1	0.958	11.93	138	1	1.06	0.94;
	1000021	1	14	8	0	0	1	0.959	13.52	138	1	1.06	0.94;
	1000022	1	10	5	0	0	1	0.97	16.08	138	1	1.06	0.94;
	1000023	1	7	3	0	0	1	1	21	138	1	1.06	0.94;
	1000024	2	13	0	0	0	1	0.992	20.89	138	1	1.06	0.94;
	1000025	2	0	0	0	0	1	1.05	27.93	138	1	1.06	0.94;
	1000026	2	0	0	0	0	1	1.015	29.71	345	1	1.06	0.94;
	1000027	2	71	13	0	0	1	0.968	15.35	138	1	1.06	0.94;
	1000028	1	17	7	0	0	1	0.962	13.62	138	1	1.06	0.94;
	1000029	1	24	4	0	0	1	0.963	12.63	138	1	1.06	0.94;
	1000030	1	0	0	0	0	1	0.968	18.79	345	1	1.06	0.94;
	1000031	2	43	27	0	0	1	0.967	12.75	138	1	1.06	0.94;
	1000032	2	59	23	0	0	1	0.964	14.8	138	1	1.06	0.94;
	1000033	1	23	9	0	0	1	0.972	10.63	138	1	1.06	0.94;
	1000034	2	59	26	0	14	1	0.986	11.3	138	1	1.06	0.94;
	1000035	1	33	9	0	0	1	0.981	10.87	138	1	1.06	0.94;
	1000036	2	31	17	0	0	1	0.98	10.87	138	1	1.06	0.94;
	1000037	1	0	0	0	-25	1	0.992	11.77	138	1	1.06	0.94;
	1000038	1	0	0	0	0	1	0.962	16.91	345	1	1.06	0.94;
	1000039	1	27	11	0	0	1	0.97	8.41	138	1	1.06	0.94;
	1000040	2	66	23	0	0	1	0.97	7.35	138	1	1.06	0.94;
	1000041	1	37	10	0	0	1	0.967	6.92	138	1	1.06	0.94;
	1000042	2	96	23	0	0	1	0.985	8.53	138	1	1.06	0.94;
	1000043	1	18	7	0	0	1	0.978	11.28	138	1	1.06	0.94;
	1000044	1	16	8	0	10	1	0.985	13.82	138	1	1.06	0.94;
	1000045	1	53	22	0	10	1	0.987	15.67	138	1	1.06	0.94;
	1000046	2	28	10	0	10	1	1.005	18.49	138	1	1.06	0.94;
	1000047	1	34	0	0	0	1	1.017	20.73	138	1	1.06	0.94;
	1000048	1	20	11	0	15	1	1.021	19.93	138	1	1.06	0.94;
	1000049	2	87	30	0	0	1	1.025	20.94	138	1	1.06	0.94;
	1000050	1	17	4	0	0	1	1.001	18.9	138	1	1.06	0.94;
	1000051	1	17	8	0	0	1	0.967	16.28	138	1	1.06	0.94;
	1000052	1	18	5	0	0	1	0.957	15.32	138	1	1.06	0.94;
	1000053	1	23	11	0	0	1	0.946	14.35	138	1	1.06	0.94;
	1000054	2	113	32	0	0	1	0.955	15.26	138	1	1.06	0.94;
	1000055	2	63	22	0	0	1	0.952	14.97	138	1	1.06	0.94;
	1000056	2	84	18	0	0	1	0.954	15.16	138	1	1.06	0.94;
	1000057	1	12	3	0	0	1	0.971	16.36	138	1	1.06	0.94;
[... 488 lines cut ...]
	'Caldwell  V2';
	'Baileysv  V2';
	'Sundial   V2';
	'Bradley   V2';
	'Hinton    V2';
	'Glen Lyn  V2';
	'Wythe     V2';
	'Smythe    V2';
	'Claytor   V2';
	'Hancock   V2';
	'Roanoke   V2';
	'Cloverdl  V2';
	'Reusens   V2';
	'Blaine    V2';
	'Franklin  V2';
	'Fieldale  V2';
	'DanRiver  V2';
	'Danville  V2';
	'Deer Crk  V2';
	'WMedford  V2';
	'Medford   V2';
	'KygerCrk  V2';
	'Corey     V2';
	'WHuntngd  V2';
};

rdzman, 2024-01-02:
The first error you encountered is unrelated to DC lines. See #209 (or #201) for the solution.

In general, there should be no issue with solving the DC OPF for a system with DC lines. However, while the `t_case9_dcline` case solves fine for the AC OPF, it appears it is infeasible for the DC OPF.

Did you remember to toggle the DC line extension on using `toggle_dcline()`? For me, your case above solves fine for both AC and DC OPF.

```
mpc = loadcase('prepared_case'); 
mpc = toggle_dcline(mpc, 'on');
results_dc = rundcopf(mpc);
results_ac = runopf(mpc);
```

yanda1234, 2024-01-02:
Hi Dr.Zimmerman,

Happy new year!
Thanks for your prompt reply.
May I ask two questions?
1. After achieving convergence in the DCOPF, I observed an anomaly in the results. Please focus on bus 4. In the original IEEE 118 case, there are two branches connecting to this bus, which is associated with a load of 39 MW. However, upon introducing a DC line from bus 30 to bus 4, the branch flows remained unchanged, leading to confusion. Do you know why?
[image]

2. In #201, you mentioned using the two attached files to replace existing files in matlab installation. Do you mean the matlab main folder, such as MATLAB_R2023b or some subfolder such as MATLAB_R2023b/bin? 

Kindly let me know your thoughts! Many thanks!

Best,
Yanda

rdzman, 2024-01-02:
1. What is the reported flow on the DC line connected to bus 4? Depending on the flow limits and the relative prices, costs, etc., the optimal solution may be to dispatch the line at zero flow.
2. It should say under your **MATPOWER** installation, not MATLAB installation. I have corrected that comment. Sorry for the confusion.

yanda1234, 2024-01-02:
Thanks, Dr.Zimmerman.

I just reinstall matpower and use the github folder to replace original one. The DCOPF result is as below.
[image]
[image]

The sum of flow at bus 4 is still not zero, but since DC flow is only 1 MW this time, I am not sure if this is due to acceptable error threshold?
Besides, I suppose Pf column specifies DC flow amount, why can this value change?
[image]

Best,
Yanda

Best,
Yanda

rdzman, 2024-01-03:
If you are running an OPF, the flow in the DC line is dispatchable. For the DC line from 30 to 4, the flow in the input appears to be 10 MW (= Pmax), but in the solved case, it is 1 MW (=Pmin).

yanda1234, 2024-01-03:
Thanks, Dr. Zimmerman. Is there anyway to fix DC flow in DCOPF? For example, MISO wants to evaluate DCOPF within its territory while keeping power interchange with neighbors, such as PJM and SPP etc.

rdzman, 2024-01-03:
Certainly.

The DC OPF finds the optimal dispatch for the line between the lower and upper limits (Pmin and Pmax), so if you want a fixed value, simply set Pmin = Pmax = <desired value>.

yanda1234, 2024-01-03:
Thanks, Dr. Zimmerman!

closed 2024-01-03 by yanda1234