MATPOWER/matpower #221: HVDC lines

yanda1234, 2024-01-25:
Hi Dr. Zimmerman,

I am interested in knowing whether it is permissible to exclusively utilize a DC line to sustain load or generation in an isolated bus. I conducted a test using the provided code and observed that the load is no longer present on that bus. The fundamental logic involves substituting AC line 110-112 with a DC line of sufficient capacity. Bus 112 carries a load of 68 MW. My objective is to observe the DC line transferring the 68 MW load from bus 110 to bus 112. The code is attached below. Kindly advise!

_initial_case=loadcase('case118.m');

tf_select_branch112_1=ismember(initial_case.branch(:,1),112);
tf_select_branch112_2=ismember(initial_case.branch(:,2),112);
initial_case.branch(logical(tf_select_branch112_1+tf_select_branch112_2),:)=[];

initial_case=rmfield(initial_case,'bus_name');
%    fbus    tbus    status    Pf    Pt    Qf    Qt    Vf    Vt    Pmin    Pmax    QminF    QmaxF    QminT    QmaxT    loss0    loss1
initial_case.dcline = [
     30    4    1    10    8.9    0    0    1.01    1    10    10    -10    10    -10    10    1    0.01;
     110   112    1    10    9.5    0    0    1    0.98    0    100    -10    10    -10    10    0    0.05;
 ];

 %% DC line cost data
 %    1    startup    shutdown    n    x1    y1    ...    xn    yn
 %    2    startup    shutdown    n    c(n-1)    ...    c0
 initial_case.dclinecost = [
     2    0    0    2    0    0    0    0    0    0    0    0    0    0;
     2    0    0    2    7.3    0    0    0    0    0    0    0    0    0;
 ];
initial_case2 = toggle_dcline(initial_case, 'on');
results_dc = rundcopf(initial_case2);_

Best,
Yanda

rdzman, 2024-01-25:
The DC line tests in `t_dcline` include one with an isolated generator bus connected only by a DC line ...

https://github.com/MATPOWER/matpower/blob/2028decd8c503c85578f2f53375ac73b32ac5cf2/lib/t/t_dcline.m#L279-L311

I suspect it should work with a load bus as well, but haven't attempted your case yet.

yanda1234, 2024-01-25:
Hi Dr. Zimmerman,

Thank you for your reply. Upon revisiting the results of my case, I observed that when bus 112 is connected with a DC line, it appears as though an additional generator has been added to it, and the dispatched power matches its load. Can you provide insights into the potential reasons behind this occurrence?
[image]

Best,
Yanda

rdzman, 2024-01-25:
There is already a generator at bus 112 in the original model. It just isn't dispatched in the original, because power from the rest of the grid is cheaper. When you put the DC line there and disconnect the AC line, the cost of the local generation is cheaper than importing via the DC line, so the local generator is dispatched, not the DC line. If you turn that generator off, the DC line provides the expected 68 MW.

```
define_constants
mpc = loadcase('case118.m');
il = find(mpc.branch(:, F_BUS) == 112 | mpc.branch(:, T_BUS) == 112);
mpc.branch(il, BR_STATUS) = 0;
ig = find(mpc.gen(:, GEN_BUS) ==  112);
mpc.gen(ig, GEN_STATUS) = 0;
mpc.bus(112, BUS_TYPE) = REF;

% fbus tbus status Pf Pt Qf Qt Vf Vt Pmin Pmax QminF QmaxF QminT QmaxT loss0 loss1
mpc.dcline = [
30 4 1 10 8.9 0 0 1.01 1 10 10 -10 10 -10 10 1 0.01;
110 112 1 10 9.5 0 0 1 0.98 0 100 -10 10 -10 10 0 0.05;
];

%% DC line cost data
% 1 startup shutdown n x1 y1 ... xn yn
% 2 startup shutdown n c(n-1) ... c0
mpc.dclinecost = [
2 0 0 2 0 0;
2 0 0 2 7.3 0;
];

mpc = toggle_dcline(mpc, 'on');
results_dc = rundcopf(mpc);
```

FWIW, decreasing the cost on the DC line to zero and leaving the generator on results in some of the load being served by the local generator and some by the DC line.

yanda1234, 2024-01-25:
Thanks, Dr. Zimmerman! Sorry for this silly mistake.

closed 2024-01-25 by yanda1234

rdzman, 2024-01-25:
No worries.