MATPOWER/matpower #224: How to warm start OPF?

viktor-ktorvi, 2024-02-13:
Hi,

I'm trying to figure out how to warm start an OPF solution. I though I'd just set the `Va`, `Vm`, `Pg` and `Qg` fields and the solver would run much faster, however, I found that whichever way I init, the solve time remains roughly the same e.g., whether I init flat, or with a solved OPF or completely at random, the time is the same. What am I doing wrong?

```
Solve time with flat init = 0.5249097347259521 sec; Iterations: 21; Converged = True
Solve time with warm init = 0.5100247859954834 sec; Iterations: 21; Converged = True
Solve time with random init = 0.5253119468688965 sec; Iterations: 21; Converged = True
```

I'm doing this in python with PYPOWER, I assume it's the same. I asked in the PYPOWER github but I didn't get an answer. I see this repo is much more active. I apologize for the confusion. I've provided the simple code that reproduces the behavior below.

<details>

<summary>Click for code</summary>

```python
from typing import Dict

import numpy as np
import pandapower as pp
import pandapower.networks as pn
from pypower import idx_bus, idx_gen
from pypower.ppoption import ppoption
from pypower.runopf import runopf

def print_solution_info(solved_ppc: Dict, name: str):
    print(f"Solve time with {name} = {solved_ppc['et']} sec; Iterations: {solved_ppc['raw']['output']['iterations']}; Converged = {solved_ppc['success']}")

def warm_start(base_ppc: Dict, warm_ppc: Dict) -> Dict:
    base_ppc["bus"][:, idx_bus.VM] = warm_ppc["bus"][:, idx_bus.VM]
    base_ppc["bus"][:, idx_bus.VA] = warm_ppc["bus"][:, idx_bus.VA]

    base_ppc["gen"][:, idx_gen.PG] = warm_ppc["gen"][:, idx_gen.PG]
    base_ppc["gen"][:, idx_gen.QG] = warm_ppc["gen"][:, idx_gen.QG]

    return base_ppc

def random_start(base_ppc: Dict) -> Dict:
    bus_size = base_ppc["bus"].shape[0]
    base_ppc["bus"][:, idx_bus.VM] = np.random.randn(bus_size)
    base_ppc["bus"][:, idx_bus.VA] = np.random.randn(bus_size)

    gen_size = base_ppc["gen"].shape[0]
    base_ppc["gen"][:, idx_gen.PG] = np.random.randn(gen_size)
    base_ppc["gen"][:, idx_gen.QG] = np.random.randn(gen_size)

    return base_ppc

def main():
    ppopt = ppoption(OUT_ALL=0, VERBOSE=0)

    net = pn.case118()
    base_ppc = pp.converter.to_ppc(net, init="flat")

    solved_flat = runopf(base_ppc, ppopt=ppopt)
    print_solution_info(solved_flat, "flat init")

    warm_started_ppc = warm_start(base_ppc, solved_flat)
    solved_warm = runopf(warm_started_ppc, ppopt=ppopt)
    print_solution_info(solved_warm, "warm init")

    random_started_ppc = random_start(base_ppc)
    solved_random = runopf(random_started_ppc, ppopt=ppopt)
    print_solution_info(solved_random, "random init")

if __name__ == "__main__":
    main()

```

</details>

rdzman, 2024-02-14:
I will answer only for MATPOWER, as I don't have direct experience with PYPOWER. And I assume you are talking about AC OPF. The `opf.start` option controls how MATPOWER determines the starting point for the OPF algorithm. From the help text ...

```
   opf.start               0           strategy for initializing OPF start pt
       [   0 - default, MATPOWER decides based on solver                   ]
       [       (currently identical to 1)                                  ]
       [   1 - ignore current state in MATPOWER case (only applies to      ]
       [       fmincon, Ipopt, Knitro and MIPS, which use an interior pt   ]
       [       estimate; others use current state as with opf.start = 2)   ]
       [   2 - use current state in MATPOWER case                          ]
       [   3 - solve power flow and use resulting state                    ]
```

The effectiveness of option 2 depends greatly on the solver. For solvers that use barrier methods like MIPS (and probably fmincon, IPOPT and Artelys Knitro), even when starting at the optimal solution, the barrier pushes the solution away from any binding constraints and it take a number of iterations for the barrier parameter to be reduced and for the algorithm to converge. I believe IPOPT and Knitro may have mechanisms to explicitly help with warm starts, but I do not have experience with them. Unfortunately, MIPS does not implement any explicit warm start assistance, so the improvement from using `mpoption('opt.start', 2)` is not overwhelming.

FWIW, the old MINOPF solver (defunct since the switch from 32-bit to 64-bit MATLAB) was quite good at warm starts.

viktor-ktorvi, 2024-03-18:
Thanks for the great explaination! I'd like to give MINOPF a try. It's website mentions that you should be contacted to acquire a copy of the solver (Windows). I'm wondering if it will even function on current versions of MATLAB and MATPOWER since its defunct?

viktor-ktorvi, 2024-03-18:
I actually would! That'd be be great!

rdzman, 2024-03-18:
No problem.

Could you please send me the following (via direct e-mail to my Cornell address) and I’ll provide you a download link. As part of the licensing terms for MINOPF,  I am required to provide the info below for each person we distribute MINOPF to.

Platform (Linux, Mac or Windows)
Name
E-mail
Affiliation
City
State/Province
Country

closed 2024-04-16 by rdzman