MATPOWER/matpower #225: Case files with transformers that perform voltage transformation.

Abijith20, 2024-02-16:
Does the MATPOWER library contain any case files featuring transformers that perform voltage transformation? While exploring the case files, I encountered case14.m, which includes transformers, but appears not to involve voltage transformation. Are there other case files within MATPOWER that demonstrate this functionality?

rdzman, 2024-02-16:
By "voltage transformation" I assume you mean a transformer that connects two buses with different values for `basekV`. Unfortunately, it looks like the `basekV` values in `case14` are all 0 so we just don't have that information.

I'm afraid I don't know about other cases off the top of my head, but it should be simple to write a script to for any branch that connects buses with different `basekV` ... those branches should be transformers that are doing "voltage transformation".

Abijith20, 2024-02-16:
%% system MVA base
mpc.baseMVA = 1;

%% bus data
%	bus_i	type	Pd	Qd	Gs	Bs	area	Vm	Va	baseKV	zone	Vmax	Vmin
mpc.bus = [
	1	3	0	0	0	0	1	1	0	66	1	1.1	0.9;
	2	1	0.4	0.2	0	0	1	1	0	11	1	1.1	0.9;
	3	1	0.4	0.2	0	0	1	1	0	11	1	1.1	0.9;
	4	1	0.4	0.2	0	0	1	1	0	11	1	1.1	0.9;
];

%% generator data
%	bus	Pg	Qg	Qmax	Qmin	Vg	mBase	status	Pmax	Pmin	Pc1	Pc2	Qc1min	Qc1max	Qc2min	Qc2max	ramp_agc	ramp_10	ramp_30	ramp_q	apf
mpc.gen = [
	1	0	0	10	-10	1.05	100	1	10	0	0	0	0	0	0	0	0	0	0	0	0;
];

%% branch data
%	fbus	tbus	r	x	b	rateA	rateB	rateC	ratio	angle	status	angmin	angmax
mpc.branch = [
	2	3	0.206	0.079	0	0	0	0	0	0	1	-360	360;
	1	2	0.206	0.079	0	0	0	0	1	0	1	-360	360;
	4	3	0.206	0.079	0	0	0	0	0	0	1	-360	360;
];

Does this script mean that I have a transformer between bus 1 and 2 which transforms the voltage from 66 kV to 11 kV?

closed 2024-04-16 by rdzman