MATPOWER/matpower #227: How to model a flexibility providing units in the matpower case file

akhtar3100, 2024-03-21:
My question is how to model flexible providing units (loads and generation) with different P-Q capability shapes in matpower.
For example, if I have a load with active and reactive power values of (50,20) in mpc.bus. What if I want to change these value not rectangularly but in other shape for example have P between 0 and 20 MW, Q between 0 and 5 MVar and P between 20 to 50 Q between  10 to 20 MVAR. How can I do it within the case file and how can i do outside of case file? same for generators.

rdzman, 2024-03-21:
I'm going to assume this is in the context of an OPF.

First, let's talk about generators. I assume you already know about the trapezoidal capability curves that are built in to MATPOWER (see Section 6.4.3 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf)). There is no reason the shape has to be like the one shown in Figure 6-6, with the reactive capability decreasing with increasing output. You simply have the option to define two sloped lines that constrain the capability curve along with the box defined by PMIN-PMAX-QMIN-QMAX.

However, that doesn't sound exactly like what you describe. So, if I understand what you want, I believe you can accomplish it by modeling your generator as two separate generators, with different reactive ranges. You will want to use a slightly lower active power cost on the first generator to ensure that it get's dispatched first. _Although, now that I think about it, if reactive power needs are great it could still choose the 2nd unit before the first if the active power cost difference is not great enough._

For the loads, you can simply treat them as negative generators, i.e. move them from the `bus` matrix to the `gen` matrix. The [`load2disp()`](https://matpower.org/docs/ref/matpower7.1/lib/load2disp.html) function can help you with this. However, there is one important caveat here. MATPOWER treats any generator with PMIN < 0 and PMAX == 0 as a special case dispatchable load **with a constant power factor constraint added** (see Section 6.4.2 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf)). In your case, you do **not** want this constant power factor constraint. You can eliminate it by setting PMAX on these generators to a negative number (e.g. `-eps`).

akhtar3100, 2024-03-27:
Static generators are model as negative loads right? what if I want to have static generators and then put different kind of P_Q region on top of them. How to do that?

rdzman, 2024-03-27:
> ... for constant power factor dispatchable load how the powerfactor value is determined. how to provide constant powerfactor in that case.

It is determined by the ratio of `PMIN` and either `QMIN` or `QMAX`, whichever is non-zero. _(At least one of them must be zero)_.

>Second when we don't want constant power factor constraint for example and want to have a circular P-Q region for dispatchable loads how to do that.

MATPOWER does not implement a circular PQ region. For that you would need to implement your own non-linear constraint See Section 6.3.2 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf).

rdzman, 2024-03-27:
> Furthermore, Static generators are model as negative loads right?

They can be modeled as negative loads (in the `PD` and `QD` fields of the `bus` matrix), or they can be modeled as generators with `PMIN = PMAX = PG` and `QMIN = QMAX = QG`.

> what if I want to have static generators with different Q capabilities constant power factor, constant Q mode, how to model that?

In the context of an OPF, you simply define the "static" generator's active and reactive power limits based on the capabilities you desire. For a "static generator" I assume the active power output is fixed. If power factor is fixed, then so is the reactive output (`QMIN = QMAX`).

akhtar3100, 2024-03-28:
> It is determined by the ratio of `PMIN` and either `QMIN` or `QMAX`, whichever is non-zero. _(At least one of them must be zero)_.
> bus	Pg	Qg	Qmax	Qmin	Vg	mBase	status	Pmax	Pmin	Pc1	Pc2	Qc1min	Qc1max	Qc2min	Qc2max	ramp_agc	ramp_10	ramp_30	ramp_q	apf
5	-0.02	-0.005	0	-0.005	1	100	1	0	-0.02	0	0	0	0	0	0	Inf	Inf	Inf	Inf	0
So far this case for a dispatchable load Pg=-0.02, Qg=-0.005, Qmax=0, Qmin=-0.005, Pmin=-0.02 would powerfactor be pf=Pmin/Qmin=-0.02/0.05=0.4, but that does not make sense normally Q/P is tan(theta) not p.f so P/Q would be tan-1(theta)? Can you confirm this calculation
> MATPOWER does not implement a circular PQ region. For that you would need to implement your own non-linear constraint See Section 6.3.2 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf).
ok

akhtar3100, 2024-03-28:
> They can be modeled as negative loads (in the `PD` and `QD` fields of the `bus` matrix), or they can be modeled as generators with `PMIN = PMAX = PG` and `QMIN = QMAX = QG`.
>
>
> In the context of an [...]

What if I want to set powerfactor limits for the static generator for powerflow? What I want to do is that I want to have P-Q boundaries defined for static generators and then randomly choose (p,q) points within this boundary for powerflows.

rdzman, 2024-03-28:
> I think the power factor should be calculated as pf=P/(P^2+Q^2), wht do you think?**

Sorry if I was unclear. I did not mean that the power factor is **equal to** `PMIN/QMIN`, simply that the values of `PMIN` and `QMIN`, particularly their ratio and the implied angle, **define** the power factor.

rdzman, 2024-03-28:
> What if I want to set powerfactor limits for the static generator for powerflow? What I want to do is that I want to have P-Q boundaries defined for static generators and then randomly choose (p,q) points within this boundary for powerflows.

In this case P and Q are fixed within the context of the power flow computation, which does not need to know anything about the P-Q boundaries. You simply use the process you describe to select the input values for the power flow. This process **does** need to know about the P-Q boundaries. Right?

akhtar3100, 2024-03-28:
yeah. Right. I will create a process which defines these boundaries and then run power flow. I thought maybe there was some function to do it.

Thanks