MATPOWER/matpower #229: Out Of Bound Error

sudo-ac, 2024-03-26:
Hi,

I am preparing a workshop for power flow analysis. In some example cases, I always encounter this error:

Newton's method converged in 4 iterations.
error: ij(_,1): out of bound 0 (dimensions are 0x0)
error: called from
    update_z at line 78 column 21
    convert_x_m2n at line 28 column 20
    network_model_x_soln at line 220 column 47
    network_model_x_soln at line 419 column 16
    network_model_x_soln at line 161 column 16
    network_model_update at line 424 column 13
    run at line 130 column 28
    runpf at line 164 column 5

To check, I'll provide you with a simple case file:

function mpc = case3
% The network consists of 3 buses, 3 branches, and 1 generator.
%
% Network diagram:
%
%    1 ---- 2 ---- 3
%
% Generator at Bus 1.
%

mpc.version = '2';
mpc.baseMVA = 100;

%% bus data
%	bus_i	type	Pd	Qd	Gs	Bs	area	Vm	Va	baseKV	zone	Vmax	Vmin
mpc.bus = [
 1 1 0 0 0 1 1 0.8 0 0.8 0 1.2 0.8;
 2 3 0 0 0 1 1 1 0 1 0 1.2 0.8;
 3 2 0 0 0 1 1 1 0 1 0 1.2 0.8;
];

%% generator data
%	bus	Pg	Qg	Qmax	Qmin	Vg	mBase	status	Pmax	Pmin	Pc1	Pc2	Qc1min	Qc1max	Qc2min	Qc2max	ramp_agc	ramp_10	ramp_30	ramp_q	apf
mpc.gen = [
 1  1.0 0.5 0.5 -0.3  0.8 100 1 0	0	0	0	0	0	0	0	0	0	0	0	0;
];

%% branch data
%	fbus	tbus	r	x	b	rateA	rateB	rateC	ratio	angle	status	angmin	angmax
mpc.branch = [
 1 2 0.1 0.2 0.03 0 0 0 0 0 1 -360 360;
 2 3 0.2 0.3 0.04 0 0 0 0 0 1 -360 360;
 3 1 0.15 0.25 0.02 0 0 0 0 0 0 -360 360;
];

I´m working with Matpower 8.0b1

Thank you

rdzman, 2024-03-26:
Sorry it didn't give you a more useful error message. 

I believe the issue is that you have both a reference bus (bus 2 is type 3) and PV bus (bus 3 is type 2) with no generator. If you change the 2nd column of the bus matrix to `[3;1;1]` it solves fine. Though it is a very strange system, since it has zero load.

sudo-ac, 2024-03-26:
Thank you, I will try it soon. It's a strange system because I want to use it for analysis.

sudo-ac, 2024-03-26:
I added at Bus 3 a Generator (+ Bustype to 3) and changed Bus 1 and Bus 2 to Type 1. Thank you for helping.

closed 2024-03-26 by sudo-ac