MATPOWER/matpower #230: How to add multiple wind generation scenarios with affine police?

bmmddzgj, 2024-04-09:
I want to solve a scenario based ac opf problem.
The scenarios are wind power forecasting errors.
The active power of conventional generators will follow a simple affine recourse rule to adjust.
For each wind power scenario, the power flow equation and inequality constraints should be satisfied.
How can I add extra constraints using matpower?

rdzman, 2024-04-09:
If I understand correctly, you have a base AC power flow or OPF solution and you would like to find the solutions for new scenarios, with different wind outputs, where all other generation is adjusted according to a set of linear constraints with a single-degree of freedom.

If that is the case, you should be able to do it by running an OPF with a set of user-defined linear constraints to restrict the generation pattern. See Section 7.1.2 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf). Be sure you set the bounds properly on any other OPF variables you may not want to change (i.e. generator voltage setpoints).

You _could_ also just use a power flow and precompute the updated generator dispatch based on the change in net load/wind and your affine recourse adjustment. The only caveat is that since you don't know the _exact_ change in losses, your adjustment will be an approximation, with the error showing up at the slack bus. But I suspect an iteration or two of adjusting it and re-running the power flow would give you a very good solution.

bmmddzgj, 2024-05-30:
Thank u. I now know that matpower has the potential to be extended with extra linear and nonlinear constraints. But I still have problem in adding extra power flow equations. I know the function to be used is
` om.add_nln_constraint('myNonlinCons', N, iseq, fcn, hess, varsets);`
As described, 
> fcn and hess are function handles and var_sets is an optional cell array of variable set names.
I want to add the following constraints with fixed $\alpha$, $p_{w,i}$, $\Delta p_{w,i}$, and $q_{w,i}$, $\Delta q_{w,i}$
[image]
$$p_{g}=p_{g}^{0}+\alpha1^{T}(\Delta p_w)\\
p_{g,i}+p_{w,i}+\Delta p_{w,i}-p_{d,i}=G_{ii} v_i^2 + \sum_{j=1,j\ne i}^N (G_{ij}v_iv_j cos(\theta_{ij})+B_{ij}v_iv_j sin(\theta_{ij}))\\
q_{g,i}+q_{w,i}+\Delta q_{w,i}-q_{d,i}=-B_{ii} v_i^2 + \sum_{j=1,j\ne i}^N (G_{ij}v_iv_j sin(\theta_{ij})-B_{ij}v_iv_j cos(\theta_{ij}))$$
where $p_{g}^{0}$ is the to-be-optimized generation in OPF problem. How can I code with matpower?

rdzman, 2024-05-30:
First I want to make sure I understand conceptually what you are trying to do. Initially, I thought you were solving different scenarios sequentially, but it seems like you want to essentially do a sort of security-constrained OPF, guarding against different wind scenarios. Let me know if I'm close ...

You start with the standard formulation of an OPF for a base case with a nominal set of fixed wind generation injections. And you would like to add additional constraints on the base case generator dispatch for a scenario with different wind injections, where the updated generation is an affine function of the base case generation and the power balance and branch flow constraints are all satisfied?

If so, what about generators that are at a lower or upper bound in the base case? Do you ignore those limits under the new wind scenario? If not, I'm afraid the problem will be infeasible. If so, I think you may need a different formulation, where each scenario has a single scalar optimization variable (not a fixed value) that shifts all generators.

Btw, a good place to become familiar with how nonlinear constraints are added to the OPF problem is in `opf_setup.m` ([here](https://github.com/MATPOWER/matpower/blob/aa8af4f5ec70e32985b0f3785029b7a9f4b84728/lib/opf_setup.m#L433) is the line that adds the power balance constraints).

bmmddzgj, 2024-06-01:
Yes, I would like to get a base generation schedule that can deal with a set of wind generation injections with a fixed affine rule.
The base case takes the wind generation injection of its expected value, and it is feasible in my case.
Thank you for recommending my the `opf_setuo.m`. I will check the file for detail.

rdzman, 2024-06-03:
FWIW, [MOST](https://matpower.org/docs/MOST-manual-1.3.pdf) (MATPOWER Optimal Scheduling Tool) already implements something similar to what you are looking for with 2 main differences.
- It uses a DC network model.
- It redispatches the generators in the different wind scenarios via cost minimization rather than a fixed affine recourse.