MATPOWER/matpower #231: OPF on the radial system offered by MATPOWER

young-hwanlee, 2024-04-30:
I ran the OPF on the radial systems offered by MATPOWER, e.g. case10ba, case51ga, case51he, case69, case70da, case74ds, case85, case94pi, case118zh, case136ma, and case141. But it's not converged except case51ga, case51he, case69, and case141. I want to successfully run the OPF on the radial systems in all cases. Could you tell me how to do?

rdzman, 2024-04-30:
The first thing I usually try when an OPF does not converge is to turn on OPF soft limits (see Section 7.6.4 in the [MATPOWER User's Manual](https://matpower.org/docs/MATPOWER-manual-7.1.pdf)) to see if there is an infeasibility.

For example ...
```
mpc = loadcase('case10ba');
mpc = toggle_softlims(mpc, 'on');
r = runopf(mpc);
```
... results in ...
```
================================================================================
|     Soft Voltage Upper Bounds                                                |
================================================================================
Bus    Voltage   Limit   Overload    mu
  #    Mag(pu)   (pu)     (pu)     ($/pu)
-----  -------  -------  -------  ---------
    1   1.052    1.000    0.052  99999.999
                        --------
               Total:      0.05
```

Here's what I found for the cases you listed by running them with OPF soft limits:
- `case10ba` - infeasible to due voltage limits
- `case51ga` - converges
- `case51he` - converges
-  `case69` - converges
- `case70da` - infeasible to due voltage limits
- `case74ds` - converges
- `case85` - infeasible to due voltage limits
- `case94pi` - crashes with soft limits enabled 😳 _(I need to look into this one!)_
- `case118zh` - infeasible to due voltage limits
- `case136ma` - infeasible to due generator limits
- `case141` - converges

rdzman, 2024-04-30:
Looks like `case94pi` has an error. It has 1 row in `gen`, but 2 in `gencost`. Deleting the duplicate row in `gencost` results in ...

- `case94pi` - infeasible due to voltage limits

I will fix that error in `case94pi` for the next release.

rdzman, 2024-05-02:
The `runopf()` function calls `ext2int()` internally. Is there a reason you need to call it explicitly beforehand?

If so, you can always remove the `order` field from `mpc` before calling `runopf()`, but that will prevent you from being able to call `int2ext()` on the `result` to get back to the original numbering. If you need to be able to do that, you'd have to save the `order` field from the `mpc` and restore it to the `result` before calling `int2ext()`.

```
mpc = ext2int(loadcase(case_num));
mpc = toggle_softlims(mpc, 'on');
mpc = rmfield(mpc, 'order');
result = runopf(mpc, mpopt);
```