MATPOWER/matpower #232: MATPOWER 8 runpf contains redundant data

yasirroni, 2024-05-29:
With the new MATPOWER 8 release, I'm noticing new structs data in simple `runpf`:

```
['version', 'baseMVA', 'bus', 'gen', 'branch', 'gencost', 'order', 'success', 'om', 'x', 'mu', 'f', 'var', 'nle', 'nli', 'qdc', 'et', 'raw']
```

It behaves like `runopf`. Does all of that data necessary? I think we should not use `om` and `raw` since we didn't solve any optimization in simple `runpf`.

___

NOTE: it broke `matpower-pip` and I know it is my faults for not reporting while it is still on 8.0b.

yasirroni, 2024-05-29:
Current `runpf` broke `save` on `octave`.

```octave
mpc = runpf()
save('mpc.mat', 'mpc')
```

Not error on `7.1`, error on `8.0`.

rdzman, 2024-05-29:
Thanks for the heads up.

MATPOWER 8 uses the new [MP-Core](https://matpower.org/doc/dev-manual/architecture.html) by default to do the calculations for `runpf()`, so it does in fact create a mathematical model object, like `runopf()` did in previous versions, for performing the solve. When using MP-Core, it returns the [task object](https://matpower.org/doc/dev-manual/task.html) and the [math model object](https://matpower.org/doc/dev-manual/math-model.html) in the `results` struct for all `run...()` functions for consistency.

If this is problematic for your application, you have a few options.

1. Turn off MP-Core completely when using legacy `runpf()`, `runcpf()`, and `runopf()` using `have_feature('mp_core', 0)`.
2. Turn off MP-Core for a specific call to legacy `runpf()`, `runcpf()`, and `runopf()` using the `exp.use_legacy_core` option, i.e.  `mpopt = mpoption('exp.use_legacy_core', 1)`.
3. Delete the objects from the `results` struct before attempting to save in Octave.
   ```
   results = runpf();
   results = rmfield(results, 'task');
   results = rmfield(results, 'om');
   save -v7 results.mat results
   ```

The first two options use the [legacy MATPOWER framework](https://matpower.org/doc/dev-manual/architecture.html#two-matpower-frameworks) and should be fully compatible with MATPOWER 7.1, but it will obviously not give access to MP-Core-only features (e.g. using `fsolve` to solve power flow).

*Note:* You should also always be able to save a solved case using `savecase()` rather than saving the raw `results` struct.

Thanks again for the heads up, and let me know if you run into other issues with the transition to MATPOWER 8.

yasirroni, 2024-06-04:
Thank you for your answer. It is to support backward compatibility and seamless communication between Python and Octave. Sadly, the `save` command is not to actually save the data, but communication between two programming language via `.MAT` file.

Currently, I'm working on oct2py capability to skip data that can't be converted to `.MAT` file, as you suggested in `Delete the objects from the results struct before attempting to save in Octave.`.

Next step is the ability to temporarily store `task` and `om` in `octave` and pass the pointer only to Python.

If you have any suggestion, feel free to contact me.

___

But, I have a question. What is it the main downside in using `have_feature('mp_core', 0)`, in using `runpf` and `runopf`? If you have any article or doc about the drawback, feel free to share it to me (as it will be highleighted in matpower python port).

rdzman, 2024-06-04:
At the moment, when using the legacy MATPOWER framework (i.e. `runpf()`, `runcpf()`, `runopf()`, etc.), there is very little downside to using the legacy core, as opposed to MP-Core. There are very few legacy framework features that are available only when using MP-Core. A few that come to mind are the ability to use `fsolve()` as an AC power flow solver, and the ability to do continuation power flow using cartesian coordinate voltage representation and/or nodal current balance.

The real difference comes when using the new flexible framework (i.e. `run_pf()`, `run_cpf()`, `run_opf()` ... _note the underscores_). I expect all new development going forward will be focused on the flexible framework ... three-phase unbalanced modeling, MATPOWER extensions, other modeling enhancements, etc.

yasirroni, 2024-06-05:
Thanks. I will try to read more about all of that functionality and how to best support matpower-pip.

In the mean time, skip unsavable octave data is proposed in this PR https://github.com/blink1073/oct2py/pull/321.

closed 2024-06-05 by yasirroni

yasirroni, 2024-06-09:
The PR is merged in oct2py https://github.com/blink1073/oct2py/pull/321#pullrequestreview-2106308842

Now, MATPOWER 8 should be supported natively by oct2py and matpower-pip. Either install from master at GitHub or wait pypi release of new oct2py.

Thanks.